ros2/rmw_fastrtps #700: fastrtps (Fast-DDS) version 2.3.x does not have ContentFilteredTopic required by humble branch

henriksod, 2023-07-27:
## Bug report

**Required Info:**

- Version or commit hash:
  - c6efdc9fd47cd50043535a2bff27a5c402728bab
 
#### Steps to reproduce issue

Incorrect fastrtps version is listed in the humble branch of this repo. It is listed as `2.3`:
https://github.com/ros2/rmw_fastrtps/blob/c6efdc9fd47cd50043535a2bff27a5c402728bab/rmw_fastrtps_shared_cpp/CMakeLists.txt#L47

fastrtps `2.3.x` does not have `fastdds/dds/topic/ContentFilteredTopic.hpp`, as is required here:
https://github.com/ros2/rmw_fastrtps/blob/c6efdc9fd47cd50043535a2bff27a5c402728bab/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L32

This feature was added in fastrtps `2.5.0`.

#### Expected behavior

The correct minimum fastrtps version to be listed in `rmw_fastrtps_shared_cpp/CMakeLists.txt`

#### Actual behavior

An incompatible minimum version of fastrtps is currently listed under `rmw_fastrtps_shared_cpp/CMakeLists.txt` in the humble branch.

#### Implementation considerations

Replace `2.3` L47 and L48 with `2.5`:
https://github.com/ros2/rmw_fastrtps/blob/c6efdc9fd47cd50043535a2bff27a5c402728bab/rmw_fastrtps_shared_cpp/CMakeLists.txt#L47

clalancette, 2023-07-27:
Please feel free to open a pull request to update that; it would be much appreciated.

fujitatomoya, 2023-07-28:
> Replace 2.3 L47 and L48 with 2.7.1

https://github.com/eProsima/Fast-DDS/pull/2825 has been backported to `2.6.x` branch with https://github.com/eProsima/Fast-DDS/pull/2843, which is available on `2.6.2` or later.

So i think it should be update into `2.6.2`.

fujitatomoya, 2023-10-09:
This has been addressed with Humble patch release https://github.com/ros2/rmw_fastrtps/commit/4b0f30f15fad0e1a9824db0bf54dbbaa5085fa5d

```bash
root@tomoyafujita:~# dpkg -l | grep rmw-fastrtps
ii  ros-humble-rmw-fastrtps-cpp                       6.2.4-1jammy.20230919.211331            amd64        Implement the ROS middleware interface using eProsima FastRTPS static code generation in C++.
ii  ros-humble-rmw-fastrtps-shared-cpp                6.2.4-1jammy.20230919.205954            amd64        Code shared on static and dynamic type support of rmw_fastrtps_cpp.
```

and as we described above, it uses, `v2.6.2`

https://github.com/ros2/rmw_fastrtps/blob/4b0f30f15fad0e1a9824db0bf54dbbaa5085fa5d/rmw_fastrtps_shared_cpp/CMakeLists.txt#L47-L48

I will go ahead to close this issue.

closed 2023-10-09 by fujitatomoya