ros2/rmw_fastrtps #703: String comparison with equal-to operator does not work for c strings, in type_support_common.cpp

henriksod, 2023-07-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - source
- Version or commit hash:
  - 6.2.3
- DDS implementation:
  - rmw_fastrtps_dynamic_cpp

#### Steps to reproduce issue
This problem was discovered in and affected this PR https://github.com/mvukov/rules_ros2/pull/148

Got this error:
```bash
'Unknown typesupport identifier, at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_registry.cpp:93'
```

Added some prints here:
```cpp
type_support_ptr TypeSupportRegistry::get_message_type_support(
  const rosidl_message_type_support_t * ros_type_support)
{
  auto creator_fun = [&ros_type_support]() -> type_support_ptr
    {
      std::cerr << ros_type_support->typesupport_identifier << "\n";
      std::cerr << "c "
        << using_introspection_c_typesupport(ros_type_support->typesupport_identifier)
        << " " << rosidl_typesupport_introspection_c__identifier <<  "\n";
      std::cerr << "cpp "
        << using_introspection_cpp_typesupport(ros_type_support->typesupport_identifier)
        << " " << rosidl_typesupport_introspection_cpp::typesupport_identifier << "\n";
      if (using_introspection_c_typesupport(ros_type_support->typesupport_identifier)) {
        auto members = static_cast<const rosidl_typesupport_introspection_c__MessageMembers *>(
          ros_type_support->data);
        return new MessageTypeSupport_c(members, ros_type_support);
      } else if (using_introspection_cpp_typesupport(ros_type_support->typesupport_identifier)) {
        auto members = static_cast<const rosidl_typesupport_introspection_cpp::MessageMembers *>(
          ros_type_support->data);
        return new MessageTypeSupport_cpp(members, ros_type_support);
      }
      RMW_SET_ERROR_MSG("Unknown typesupport identifier");
      return nullptr;
    };

  return get_type_support(ros_type_support, message_types_, creator_fun);
}
```

Outcome:
```bash
rosidl_typesupport_introspection_cpp
c 0 rosidl_typesupport_introspection_c
cpp 1 rosidl_typesupport_introspection_cpp
rosidl_typesupport_introspection_c
c 0 rosidl_typesupport_introspection_c
cpp 0 rosidl_typesupport_introspection_cpp
```

#### Expected behavior

`rosidl_typesupport_introspection_c == rosidl_typesupport_introspection_c`

#### Actual behavior

`rosidl_typesupport_introspection_c != rosidl_typesupport_introspection_c`

#### Implementation considerations

Use `strcmp`:
```cpp
bool using_introspection_c_typesupport(const char * typesupport_identifier)
{
  return !strcmp(typesupport_identifier, rosidl_typesupport_introspection_c__identifier);
}
```

or `std::string().compare`

```cpp
bool using_introspection_c_typesupport(const char * typesupport_identifier) {
  return !std::string(typesupport_identifier)
    .compare(rosidl_typesupport_introspection_c__identifier);
 }
```

Update both these lines:

https://github.com/ros2/rmw_fastrtps/blob/b45e73500bc0a2056fdc31197a6f94434b327d55/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp#L28
https://github.com/ros2/rmw_fastrtps/blob/b45e73500bc0a2056fdc31197a6f94434b327d55/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp#L34

iuhilnehc-ynos, 2023-07-31:
```
cpp 1 rosidl_typesupport_introspection_cpp   
```
It returns `1`(true), so I think it's OK.

Please notice that it's a trick to compare the `char*` address directly for performance. The [typesupport_identifier](https://github.com/ros2/rosidl/blob/7583b95e13b9b0ca22f4f166da732ac73eace945/rosidl_runtime_c/include/rosidl_runtime_c/message_type_support_struct.h#L47)(type is const char*) is initialized with  https://github.com/ros2/rosidl/blob/7cbb1161e8d3fa9aff29a481bce0e6c2b56715b5/rosidl_typesupport_introspection_cpp/resource/msg__type_support.cpp.em#L251, it's better to [compare them](https://github.com/ros2/rmw_fastrtps/blob/29050bfdb02dda7a05cd87d9311f685762ca5542/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp#L34-L35) directly, right?

iuhilnehc-ynos, 2023-07-31:
The `typesupport_identifier` in rosidl_typesupport_introspection_c is similar to above, which is initialized at https://github.com/ros2/rosidl/blob/7cbb1161e8d3fa9aff29a481bce0e6c2b56715b5/rosidl_typesupport_introspection_c/resource/msg__type_support.c.em#L287 and will be assigned with https://github.com/ros2/rosidl/blob/7cbb1161e8d3fa9aff29a481bce0e6c2b56715b5/rosidl_typesupport_introspection_c/resource/msg__type_support.c.em#L311. Could you provide more detailed steps to reproduce this issue?  A test case might be better.

henriksod, 2023-07-31:
> ```shell
> rosidl_typesupport_introspection_c
> c 0 rosidl_typesupport_introspection_c
> cpp 0 rosidl_typesupport_introspection_cpp
> ```

The second step fails:
```bash
rosidl_typesupport_introspection_c
c 0 rosidl_typesupport_introspection_c
cpp 0 rosidl_typesupport_introspection_cpp
```

My theory is that they are not pointing to the same address in memory

iuhilnehc-ynos, 2023-07-31:
I am afraid that I can't reproduce this issue by rolling in my machine.

```patch
diff --git a/rclcpp/test/rclcpp/test_generic_pubsub.cpp b/rclcpp/test/rclcpp/test_generic_pubsub.cpp
index f4cef0b7..56b74089 100644
--- a/rclcpp/test/rclcpp/test_generic_pubsub.cpp
+++ b/rclcpp/test/rclcpp/test_generic_pubsub.cpp
@@ -263,3 +263,24 @@ TEST_F(RclcppGenericNodeFixture, generic_publisher_uses_qos)
   // It normally takes < 20ms, 5s chosen as "a very long time"
   ASSERT_TRUE(wait_for(connected, 5s));
 }
+
+
+TEST_F(RclcppGenericNodeFixture,
+     WhenPublisherDeletedInThread_EnsureCleanNodeCleanup) {
+  auto node = rclcpp::Node::make_shared("pub_node");
+  auto publisher =
+      node->create_generic_publisher("pub_topic", "std_msgs/String", 10);
+
+  auto reset_publisher = std::async(std::launch::async, [&node, &publisher]() {
+    std::this_thread::sleep_for(std::chrono::milliseconds(200));
+    publisher.reset();
+
+    // New: Create a timer that gets immediately out of scope (should never
+    // fire) to trigger cleanup of the publisher.
+    node->create_wall_timer(std::chrono::seconds(1), []() {});
+  });
+
+  rclcpp::executors::SingleThreadedExecutor executor;
+  executor.add_node(node);
+  executor.spin_until_future_complete(reset_publisher);
+}
\ No newline at end of file
```

test log:

```shell
chenlh rclcpp $ RMW_IMPLEMENTATION=rmw_fastrtps_dynamic_cpp ./test_generic_pubsub__rmw_fastrtps_dynamic_cpp --gtest_filter="*.WhenPublisherDeletedInThread_EnsureCleanNodeCleanup"
Running main() from gmock_main.cc
Note: Google Test filter = *.WhenPublisherDeletedInThread_EnsureCleanNodeCleanup
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from RclcppGenericNodeFixture
[ RUN      ] RclcppGenericNodeFixture.WhenPublisherDeletedInThread_EnsureCleanNodeCleanup
[       OK ] RclcppGenericNodeFixture.WhenPublisherDeletedInThread_EnsureCleanNodeCleanup (2291 ms)
[----------] 1 test from RclcppGenericNodeFixture (2291 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (2295 ms total)
[  PASSED  ] 1 test.

```

iuhilnehc-ynos, 2023-07-31:
> This line looks interesting?

Yeah, I noticed that, but it's not a big deal. Because it will be assigned with [`rosidl_typesupport_introspection_c__identifier`](https://github.com/ros2/rosidl/blob/7cbb1161e8d3fa9aff29a481bce0e6c2b56715b5/rosidl_typesupport_introspection_c/resource/msg__type_support.c.em#L311)

henriksod, 2023-07-31:
It is being called two times:
```cpp
type_support_ptr TypeSupportRegistry::get_message_type_support(
  const rosidl_message_type_support_t * ros_type_support)
{
  auto creator_fun = [&ros_type_support]() -> type_support_ptr
    {
      std::cerr << "Got: " << &ros_type_support->typesupport_identifier << "\n";
      std::cerr << "Expect: " << &rosidl_typesupport_introspection_c__identifier << "\n";
      if (using_introspection_c_typesupport(ros_type_support->typesupport_identifier)) {
        ...
```

```bash
Got: 0x7f44da115d20
Expect: 0x7f44da115f58
Got: 0x562bf2d630d0
Expect: 0x7f44da115f58
```

Somehow there is a memory address mismatch

iuhilnehc-ynos, 2023-07-31:
I think you need to output `(void*)ros_type_support->typesupport_identifier` instead of `&ros_type_support->typesupport_identifier`, and update the two others as well.

henriksod, 2023-07-31:
Yields same result

```bash
Got: 0x7f42074c78c8 rosidl_typesupport_introspection_cpp
Expected C: 0x7f42074c78a0 rosidl_typesupport_introspection_c
Expected CPP: 0x7f42074c78c8 rosidl_typesupport_introspection_cpp
Got: 0x563a0a646b28 rosidl_typesupport_introspection_c
Expected C: 0x7f42074c78a0 rosidl_typesupport_introspection_c
Expected CPP: 0x7f42074c78c8 rosidl_typesupport_introspection_cpp
```

henriksod, 2023-07-31:
Doing some more digging,

I noticed that here `strcmp` is used:
https://github.com/ros2/rosidl/blob/cf3b637605c8c1dc0b1266ca0090963e9186c7dd/rosidl_runtime_c/src/message_type_support.c#L37

Shouldn't it use `==` as well then? What is the performance gain in practice, is this function called a lot? I suppose a ros network does not have > 10000 nodes.

clalancette, 2023-07-31:
We've had this conversation in the past, though I cannot find the issue right at the moment.  The short answer is that we have two unanswered questions here:

1.  Is the expectation that the pointers are the same correct?
2.  What is the performance impact of switching to a `strcmp` here?

If we can determine the answer to both of those, then we can make a decision about which way to go.

henriksod, 2023-07-31:
Looks like it is not the address of `const rosidl_message_type_support_t * ros_type_support` that changes, but actually `rosidl_typesupport_introspection_c__identifier`.

<details>
<summary>gdb</summary>

```bash
Thread 1 "generic_publish" hit Breakpoint 16, rmw_fastrtps_dynamic_cpp::create_publisher (participant_info=0x555555ae75c0, type_supports=0x555555a720d0 <rcl_interfaces__msg__Log__rosidl_typesupport_introspection_c__Log_message_type_support_handle>, topic_name=0x555555b96400 "/rosout", qos_policies=0x7fffffff6af0, publisher_options=0x7fffffff6b70, keyed=false, create_publisher_listener=true) at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/publisher.cpp:135
135     external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/publisher.cpp: No such file or directory.
(gdb) p rosidl_typesupport_introspection_c__identifier
$33 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p &rosidl_typesupport_introspection_c__identifier
$34 = (const char **) 0x555555a74178 <rosidl_typesupport_introspection_c.identifier>
(gdb) p rosidl_typesupport_introspection_cpp::typesupport_identifier
$35 = 0x5555558dab30 "rosidl_typesupport_introspection_cpp"
(gdb) p &rosidl_typesupport_introspection_cpp::typesupport_identifier
$36 = (const char **) 0x555555a74180 <rosidl_typesupport_introspection_cpp::typesupport_identifier>
(gdb) p type_support->typesupport_identifier
$37 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p &type_support->typesupport_identifier
$38 = (const char **) 0x555555a720d0 <rcl_interfaces__msg__Log__rosidl_typesupport_introspection_c__Log_message_type_support_handle>
(gdb) continue
Continuing.

Thread 1 "generic_publish" hit Breakpoint 17, TypeSupportRegistry::get_message_type_support (this=0x7ffff79f1ac0 <TypeSupportRegistry::get_instance()::type_registry_instance>, ros_type_support=0x555555a720d0 <rcl_interfaces__msg__Log__rosidl_typesupport_introspection_c__Log_message_type_support_handle>) at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_registry.cpp:105
105     external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_registry.cpp: No such file or directory.
(gdb) p ros_type_support->typesupport_identifier
$39 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p &ros_type_support->typesupport_identifier
$40 = (const char **) 0x555555a720d0 <rcl_interfaces__msg__Log__rosidl_typesupport_introspection_c__Log_message_type_support_handle>
(gdb) continue
Continuing.
Got: 0x5555558dab08rosidl_typesupport_introspection_c
Expected C: 0x7ffff76fcd60rosidl_typesupport_introspection_c
Expected CPP: 0x7ffff76fcd88 rosidl_typesupport_introspection_cpp

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Unknown typesupport identifier, at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_registry.cpp:103'

with this new error message:

  'create_publisher() failed to get message_type_support, at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/publisher.cpp:181'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[Thread 0x7ffff3218640 (LWP 618648) exited]
[Thread 0x7ffff52bc640 (LWP 618516) exited]
[Thread 0x7ffff4abb640 (LWP 618517) exited]
[Thread 0x7ffff42ba640 (LWP 618518) exited]
[Thread 0x7ffff5abd640 (LWP 618515) exited]
[Thread 0x7ffff3a19640 (LWP 618519) exited]
unknown file: Failure
C++ exception with description "failed to initialize rcl node: error not set, at external/ros2_rcl/rcl/src/rcl/publisher.c:116" thrown in the test body.
[  FAILED  ] TestGenericPublisher.WhenPublisherDeletedInThread_EnsureCleanNodeCleanup (252135 ms)
[----------] 1 test from TestGenericPublisher (252135 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (252135 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestGenericPublisher.WhenPublisherDeletedInThread_EnsureCleanNodeCleanup

 1 FAILED TEST
[Thread 0x7ffff62be640 (LWP 618514) exited]
[Inferior 1 (process 618392) exited with code 01]
```
</details>

henriksod, 2023-07-31:
Proof:

```cpp
bool
using_introspection_c_typesupport(const char * typesupport_identifier)
{
  bool equality = typesupport_identifier == rosidl_typesupport_introspection_c__identifier;
  return typesupport_identifier == rosidl_typesupport_introspection_c__identifier;
}
```

```bash
Thread 1 "generic_publish" hit Breakpoint 1, using_introspection_c_typesupport (typesupport_identifier=0x5555558dab08 "rosidl_typesupport_introspection_c") at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp:29
29      in external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp
(gdb) p typesupport_identifier
$8 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p rosidl_typesupport_introspection_c__identifier
$9 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p equality
$10 = false  <-----------------
```

clalancette, 2023-07-31:
So the question is: what is different about this situation?  Because we don't normally see this problem.

henriksod, 2023-07-31:
```cpp
#include <cstdint>

bool
using_introspection_c_typesupport(const char * typesupport_identifier)
{
  auto a = reinterpret_cast<std::uintptr_t>(typesupport_identifier);
  auto b = reinterpret_cast<std::uintptr_t>(rosidl_typesupport_introspection_c__identifier);
  bool equality = a == b;
  return a == b;
}
```

```bash
Thread 1 "generic_publish" hit Breakpoint 1, using_introspection_c_typesupport (typesupport_identifier=0x5555558dab08 "rosidl_typesupport_introspection_c") at external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp:35
35      in external/ros2_rmw_fastrtps/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp
(gdb) p typesupport_identifier
$1 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p rosidl_typesupport_introspection_c__identifier
$2 = 0x5555558dab08 "rosidl_typesupport_introspection_c"
(gdb) p equality
$3 = false
(gdb) p a
$4 = 93824995928840
(gdb) p b
$5 = 140737344691008
```

values resulting from cast to `std::uintptr_t` are different. `typesupport_identifier` somehow gets another address.

henriksod, 2023-07-31:
[generic_publisher_tests.tar.gz](https://github.com/ros2/rmw_fastrtps/files/12215778/generic_publisher_tests.tar.gz)

If anyone is interested to investigate further, I think I have reached my capability...

Invoke gdb with:
```bash
AMENT_PREFIX_PATH="generic_publisher_tests_launch_ament_setup" gdb generic_publisher_tests_impl
```

iuhilnehc-ynos, 2023-08-01:
It seems it's the compilation issue of `rules_ros2`.

Please see the Bazel build parameter file for `rules_ros2`.

- bazel-out/k8-dbg/bin/external/ros2_common_interfaces/std_msgs/libplugin.so-2.params

```
-Wl,--start-lib
bazel-out/k8-dbg/bin/external/ros2_rosidl/_objs/rosidl_typesupport_introspection_c/identifier.pic.o
-Wl,--end-lib
-Wl,--start-lib
bazel-out/k8-dbg/bin/external/ros2_rosidl/_objs/rosidl_typesupport_introspection_cpp/identifier.pic.o
-Wl,--end-lib
```

- bazel-out/k8-dbg/bin/ros2/test/generic_publisher_tests_impl-2.params

```
-Wl,--start-lib
bazel-out/k8-dbg/bin/external/ros2_rosidl/_objs/rosidl_typesupport_introspection_c/identifier.pic.o
-Wl,--end-lib
-Wl,--start-lib
bazel-out/k8-dbg/bin/external/ros2_rosidl/_objs/rosidl_typesupport_introspection_cpp/identifier.pic.o
-Wl,--end-lib
```

A similar demo as below
[test-global-variable.zip](https://github.com/ros2/rmw_fastrtps/files/12223759/test-global-variable.zip)

```
$ ./a.out 
0x55ad13ad4078 aaaaaaaaaaa
lib: 0x55ad13ad4078 aaaaaaaaaaa
lib: 0x7f122cd01000 aaaaaaaaaaa
```

henriksod, 2023-08-01:
Yes, forcing dynamic linking of the test binary did indeed work!

closed 2023-08-01 by henriksod