ros2/rmw_fastrtps #735: add documentation how to get the fastdds log information for debugging

fujitatomoya, 2023-12-20:
## Feature request

#### Feature description

This is documentation feature request to see and print the detail log information in fastdds library.
We sometimes need to go deep inside of fastdds library information during debugging the problems and behavior even the application is running on top of ROS 2.
Using XML file configuration to change the log information provided by fastdds library would be really useful for development and issue report.

#### Implementation considerations

I think we can add the example XML file configuration to https://github.com/ros2/rmw_fastrtps?tab=readme-ov-file#example, and provide the URL link for more detailed setting as Fast-DDS.

fujitatomoya, 2023-12-20:
@Barry-Xu-2018 @iuhilnehc-ynos Do you guys happen to know this? I am willing to make this doc once i figure out how to do that.

CC: @clalancette @MiguelCompany @EduPonz

Barry-Xu-2018, 2023-12-21:
@fujitatomoya 

Do you means this document https://fast-rtps.docs.eprosima.com/en/v2.3.1/fastdds/xml_configuration/log.html#logprofiles?   
We haven't used this configuration before.

EduPonz, 2023-12-21:
This is precisely what you'd need. It is just another profile to add to your XML, as simple as that

iuhilnehc-ynos, 2023-12-21:
IIUC, the `print the detail log information` metioned by @fujitatomoya is about the INFO message.

Please notice that:

1. the [INFO](https://github.com/eProsima/Fast-DDS/blob/6898d18231b932e8d725dd2cef51dd5cbe8ffe98/include/fastdds/dds/log/Log.hpp#L345) message in the `Fast-DDS` is [OFF](https://github.com/eProsima/Fast-DDS/blob/6898d18231b932e8d725dd2cef51dd5cbe8ffe98/CMakeLists.txt#L323-L346) by default.
2. the default verbosity is [Log::Error](https://github.com/eProsima/Fast-DDS/blob/6898d18231b932e8d725dd2cef51dd5cbe8ffe98/src/cpp/fastdds/log/Log.cpp#L45)

It seems we need to re-compile the `Fast-DDS` and call the `rmw_set_log_severity` if we want to get the `the detail log`.

EduPonz, 2023-12-21:
In case you want INFO messages, then @iuhilnehc-ynos is indeed correct, you'd need to activate them at build time (see [here](https://fast-dds.docs.eprosima.com/en/latest/installation/configuration/cmake_options.html#log-options)). Mind that they are deactivated due to performance considerations.

fujitatomoya, 2023-12-21:
@Barry-Xu-2018 @iuhilnehc-ynos @EduPonz 

thanks, actually i was looking for something @iuhilnehc-ynos mentioned. (where to sink is out of scope.) sometimes we want to have the information, debug log to check the Fast-DDS internal even with ROS 2 development. Having and asking user to rebuild the Fast-DDS would be not really helpful for that case... I also understand the performance concern.

Anyway now i see that we need to rebuild the source code (and i can live with that), this was more like a question, so probably we can come back to this topic if necessary. closing this for now.

closed 2023-12-21 by fujitatomoya

EduPonz, 2023-12-22:
With recent colcon versions is possible to use Fast DDS from an overlay (before colcon had issues with this), so Fast DDS can be put in the application workspace and be built with log info when necessary

fujitatomoya, 2023-12-22:
@EduPonz yeah understood, but user needs to build the source code. it would be probably nicer if user can configure the log level at the runtime.