ros2/rmw_fastrtps #742: create_publisher() failed to get message_type_support

thomasmoore-torc, 2024-02-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Built from source
- Version or commit hash:
  - 7.1.2
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

We are augmenting https://github.com/mvukov/rules_ros2 to support `rmw_fastrtps`, which builds ROS2 using Bazel. One of the principles of Bazel is that libraries are generally statically linked. As such, constants such as `rosidl_typesupport_introspection_c__identifier` which would normally be shared between the introspection and fastrtps libraries are duplicated between them. This results in checks like `typesupport_identifier == rosidl_typesupport_introspection_c__identifier` always return `false` because the string pointers are not equivalent even though the contents of the strings are equivalent. 

While the comparisons in [rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp](https://github.com/ros2/rmw_fastrtps/blob/rolling/rmw_fastrtps_dynamic_cpp/src/type_support_common.cpp) work for the way in with `rmw_fastrtps` is traditionally compiled, they fail when the assumption that the identifier strings are globally static and shared. As such, it might be worth considering using `strcmp` to compare the identifiers instead of pointer comparison:

```diff
 bool
 using_introspection_c_typesupport(const char * typesupport_identifier)
 {
-  return typesupport_identifier == rosidl_typesupport_introspection_c__identifier;
+  return (strcmp(typesupport_identifier,
+                 rosidl_typesupport_introspection_c__identifier) == 0);
 }
 
 bool
 using_introspection_cpp_typesupport(const char * typesupport_identifier)
 {
-  return typesupport_identifier ==
-         rosidl_typesupport_introspection_cpp::typesupport_identifier;
+  return (strcmp(typesupport_identifier,
+                 rosidl_typesupport_introspection_cpp::typesupport_identifier) == 0);
 }
```

#### Expected behavior

The message type support should be acquired.

#### Actual behavior

The message type support is not acquired because the identifiers contain the same strings but are at different memory locations.

#### Additional information

fujitatomoya, 2024-02-26:
@thomasmoore-torc i may be missing something here, could you give me an example code?

> This results in checks like typesupport_identifier == rosidl_typesupport_introspection_c__identifier always return false because the string pointers are not equivalent even though the contents of the strings are equivalent.

is this true?

```cpp
#include <iostream>
#include <cstring>

static const char* str1 = "Hello";
static const char* str2 = "Hello";

int main() {
  std::cout << "Comparing with std::strcmp ----------" << std::endl;
  int result = std::strcmp(str1, str2);
  if (result == 0) {
    std::cout << "Strings are equal." << std::endl;
  } else if (result < 0) {
    std::cout << "String 1 is less than String 2." << std::endl;
  } else {
    std::cout << "String 1 is greater than String 2." << std::endl;
  }

  std::cout << "Comparing with equal operation ----------" << std::endl;
  if (str1 == str2) {
    std::cout << "Strings are equal." << std::endl;
  } else {
    std::cout << "Strings are NOT equal." << std::endl;
  }

  return 0;
}
```

and,

```bash
tomoyafujita@~/DVT/samples/C_C++/Basic >g++ strcmp.cpp
tomoyafujita@~/DVT/samples/C_C++/Basic >./a.out 
Comparing with std::strcmp ----------
Strings are equal.
Comparing with equal operation ----------
Strings are equal.
```

thomasmoore-torc, 2024-02-26:
@fujitatomoya - your example is flawed in that the two strings are combined by the compiler into a single static constant as can be seen when looking at the compiler output:

https://godbolt.org/z/soeMv3Wqq

```
.LC0:
        .string "Hello"
str1:
        .quad   .LC0
str2:
        .quad   .LC0
```

In the specific case to which I'm referring, the situation is slightly different. In a standard colcon build, the string constants (e.g. `rosidl_typesupport_introspection_c__identifier` and `rosidl_typesupport_introspection_cpp::typesupport_identifier`) are defined in the `rosidl_typesupport_introspection` libraries and the message typesupport libraries then reference that symbol:

```
$ nm ./rosidl_typesupport_introspection_c/lib/librosidl_typesupport_introspection_c.so | grep rosidl_typesupport_introspection_c__identifier
0000000000004020 D rosidl_typesupport_introspection_c__identifier
$ nm ./std_msgs/lib/libstd_msgs__rosidl_typesupport_introspection_c.so | grep rosidl_typesupport_introspection_c__identifier
                 U rosidl_typesupport_introspection_c__identifier
$ nm ./rosidl_typesupport_introspection_cpp/lib/librosidl_typesupport_introspection_cpp.so | grep typesupport_identifier
0000000000004020 D _ZN36rosidl_typesupport_introspection_cpp22typesupport_identifierE
$ nm ./std_msgs/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so | grep typesupport_identifier
                 U _ZN36rosidl_typesupport_introspection_cpp22typesupport_identifierE
```

Because the typesupport libraries are using those symbols directly, directly comparing the string pointers works in the same manner as your example because the pointers are the same. However, if the `rosidl_typesupport_introspection` libraries were to be statically linked into the message typesupport libraries, then each would have a unique copy of the identifier string, making the pointer comparison fail since the comparison would be between two completely different copies of the same string.

Unfortunately, I can't easily find a way to demonstrate this issue with a standard colcon build of ROS 2. However, I do believe that it's risky to assume that the identifier strings will always be at the the same location in memory.

fujitatomoya, 2024-02-26:
okay compiler tries to use the const string addresses, i can confirm this with object code even with local pointer.
we can make this happen with heap memory as below, i think that is the case you are saying.

```cpp
#include <iostream>
#include <cstring>

static const char* str1 = "Hello";
//static const char* str2 = "Hello";

int main() {
#if 0
  // compiler tries to optimize and attach the same pointer to it.
  const char* str2 = "Hello";
#else
  char *str2 = (char*) malloc((sizeof(str1)+1) * sizeof(char));
  (void) strcpy(str2, "Hello");
#endif
  printf("str2 is %s\n", str2);
  std::cout << "Comparing with std::strcmp ----------" << std::endl;
  int result = std::strcmp(str1, str2);
  if (result == 0) {
    std::cout << "Strings are equal." << std::endl;
  } else if (result < 0) {
    std::cout << "String 1 is less than String 2." << std::endl;
  } else {
    std::cout << "String 1 is greater than String 2." << std::endl;
  }

  std::cout << "Comparing with equal operation ----------" << std::endl;
  if (str1 == str2) {
    std::cout << "Strings are equal." << std::endl;
  } else {
    std::cout << "Strings are NOT equal." << std::endl;
  }

  return 0;
}
```

and,

```bash
tomoyafujita@~/DVT/02_TEST/cpp_strcmp >./a.out 
str2 is Hello
Comparing with std::strcmp ----------
Strings are equal.
Comparing with equal operation ----------
Strings are NOT equal.
```

i will borrow your code and fix the all rmw implementations if they need to be fixed.

fujitatomoya, 2024-02-26:
@thomasmoore-torc it would be really appreciated if you can review these.

- https://github.com/ros2/rmw_fastrtps/pull/744
- https://github.com/ros2/rmw_cyclonedds/pull/481

https://github.com/ros2/rmw_connextdds does not have related code.

thomasmoore-torc, 2024-02-28:
@fujitatomoya - those changes look good to me.

I've encountered a related issue within `rosidl_typesupport_c` and `rosidl_typesupport_cpp` at the following locations:

* https://github.com/ros2/rosidl_typesupport/blob/rolling/rosidl_typesupport_c/src/type_support_dispatch.hpp#L46
* https://github.com/ros2/rosidl_typesupport/blob/rolling/rosidl_typesupport_cpp/src/type_support_dispatch.hpp#L44

Would it make sense to go ahead and make the same change there as part of this issue?

fujitatomoya, 2024-02-28:
> Would it make sense to go ahead and make the same change there as part of this issue?

yeah, i can take care of those as well, thanks for pointing out!

closed 2024-02-29 by fujitatomoya