ros2/rmw_fastrtps #758: ASYNCHRONOUS publish mode is not setting.

Krzo99, 2024-05-21:
## Bug report

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ros-humble-fastrtps/jammy 2.6.7-1jammy.20240217.034748 ?
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Following [this](https://docs.ros.org/en/humble/Tutorials/Advanced/FastDDS-Configuration.html#mixing-synchronous-and-asynchronous-publications-in-the-same-node) example, using the exactly same code and provided .xml, and env:
```
FASTRTPS_DEFAULT_PROFILES_FILE=/my_path/SyncAsync.xml
RMW_FASTRTPS_USE_QOS_FROM_XML=1
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

Create a publisher and subscriber.
Code compiles, and works. However, if I check in [FastDDS Monitor](https://www.eprosima.com/index.php/products-all/eprosima-fast-dds-monitor), the publishing mode is still set to "SYNCHRONOUS"

[image]

#### Expected behavior

The publishing mode would be correctly set to ASYNCHRONOUS, as provided in .xml config file.

#### Actual behavior

The publishing mode is always set to SYNCHRONOUS, regardless of .xml config

#### Additional information

The SyncAsync.xml file is found correctly, as changing any tags gives errors when running the node.

Mario-DL, 2024-05-21:
Hi @Krzo99 
Thanks for the report. The issue has to do with the fact that `Fast DDS Monitor` is not reporting `ASYNCHRONOUS`, but it does not mean that the `publish_mode` is not correctly set with the configuration you provide. We will come back when we have a fix. 
In the meantime I am feeling free to correct the title so that it is more accurate.

clalancette, 2024-05-21:
Given that this looks like a bug in Fast-DDS Monitor, and not in `rmw_fastrtps`, I'm going to go ahead and close this.  Thanks for the report.

closed 2024-05-21 by clalancette