ros2/rmw_fastrtps #760: Asan regression between Humble and Iron release

luca-della-vedova, 2024-05-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binary, from [docker image](https://hub.docker.com/layers/osrf/ros/iron-desktop-full-jammy/images/sha256-780cd47de4324cacec48a12a4067b59a34ab826f842d2c15b6a624fca7d5a8d9?context=explore)
- Version or commit hash:
  - Latest binary as of time of writing
- DDS implementation:
  - fastrtps
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

The regression happened in our automated CI. Specifically, looking at the result of [this action](https://github.com/open-rmf/rmf_ci_templates/pull/10/commits/31ffe5b5b1a9dbae7671297efae5d430d5335844), it is green on Humble, and fails on Iron.

To reproduce, either fork the [rmf_ci_templates repo](https://github.com/open-rmf/rmf_ci_templates/) to run in a github action (note that we set the rmf to cyclonedds to circumvent this issue, this should be reverted) or reproduce locally by:

* Prepare an Ubuntu 22.04 machine with Iron.
* Install Open-RMF from source using the `iron` branches https://github.com/open-rmf/rmf/tree/iron.
* Build and run an address sanitizer: `colcon build --packages-up-to rmf_fleet_adapter --mixin asan-gcc lld`
* Test: `colcon test --packages-up-to rmf_fleet_adapter`.

#### Expected behavior

CI succeeds on both iron and humble.

#### Actual behavior

CI fails on iron. I haven't tried but I suspect that building the same branches on `humble` would actually work.

#### Additional information

The error reported by `asan` is reported in the PR but I'll paste it here in case it becomes unavailable:

```
2024-05-24T20:17:09.6645509Z     cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'
2024-05-24T20:17:09.6646646Z     Fail in delete datareader, at ./src/rmw_service.cpp:100 during '__function__'
2024-05-24T20:17:09.6647476Z     AddressSanitizer:DEADLYSIGNAL
2024-05-24T20:17:09.6648038Z     =================================================================
2024-05-24T20:17:09.6649235Z     ==44942==ERROR: AddressSanitizer: SEGV on unknown address 0x0000000007f0 (pc 0x7ff6c375bef4 bp 0x60c0006b2dc0 sp 0x7ffff95c7048 T0)
2024-05-24T20:17:09.6650494Z     ==44942==The signal is caused by a READ memory access.
2024-05-24T20:17:09.6651220Z     ==44942==Hint: address points to the zero page.
2024-05-24T20:17:09.6651916Z     ==44942==WARNING: invalid path to external symbolizer!
2024-05-24T20:17:09.6652730Z     ==44942==WARNING: Failed to use and restart external symbolizer!
2024-05-24T20:17:09.6654030Z         #0 0x7ff6c375bef4  (/lib/x86_64-linux-gnu/libc.so.6+0x97ef4) (BuildId: 962015aa9d133c6cbcfb31ec300596d7f44d3348)
2024-05-24T20:17:09.6655747Z         #1 0x7ff6be2ef20d  (/opt/ros/iron/lib/libfastrtps.so.2.10+0x34820d) (BuildId: a17d15658847c53e8609260c3308529bf5ffa3c1)
2024-05-24T20:17:09.6657530Z         #2 0x7ff6be2f1735  (/opt/ros/iron/lib/libfastrtps.so.2.10+0x34a735) (BuildId: a17d15658847c53e8609260c3308529bf5ffa3c1)
2024-05-24T20:17:09.6659410Z         #3 0x7ff6c19ca8ed  (/opt/ros/iron/lib/librmw_fastrtps_shared_cpp.so+0x4e8ed) (BuildId: 26ffed4ec6855b417c93e7ef8977e33baa96787f)
2024-05-24T20:17:09.6661211Z         #4 0x7ff6c5a3a89b  (/opt/ros/iron/lib/librcl.so+0x2b89b) (BuildId: 06fa6dca7f44b5a0082924dbc102de5ef1754bc0)
2024-05-24T20:17:09.6662804Z         #5 0x7ff6c5a3ac52  (/opt/ros/iron/lib/librcl.so+0x2bc52) (BuildId: 06fa6dca7f44b5a0082924dbc102de5ef1754bc0)
2024-05-24T20:17:09.6664317Z         #6 0x7ff6c5c921f3  (/opt/ros/iron/lib/librclcpp.so+0x14c1f3) (BuildId: 30853bd50c7eab48c22306ef21235a4f7e60180e)
2024-05-24T20:17:09.6665933Z         #7 0x7ff6c5c734e7  (/opt/ros/iron/lib/librclcpp.so+0x12d4e7) (BuildId: 30853bd50c7eab48c22306ef21235a4f7e60180e)
2024-05-24T20:17:09.6667349Z         #8 0x7ff6c5c71fe6  (/opt/ros/iron/lib/librclcpp.so+0x12bfe6) (BuildId: 30853bd50c7eab48c22306ef21235a4f7e60180e)
2024-05-24T20:17:09.6668987Z         #9 0x7ff6c3709a55  (/lib/x86_64-linux-gnu/libc.so.6+0x45a55) (BuildId: 962015aa9d133c6cbcfb31ec300596d7f44d3348)
2024-05-24T20:17:09.6670588Z         #10 0x7ff6c5c344d6  (/opt/ros/iron/lib/librclcpp.so+0xee4d6) (BuildId: 30853bd50c7eab48c22306ef21235a4f7e60180e)
2024-05-24T20:17:09.6671688Z     
2024-05-24T20:17:09.6672109Z     AddressSanitizer can not provide additional info.
2024-05-24T20:17:09.6673848Z     SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libc.so.6+0x97ef4) (BuildId: 962015aa9d133c6cbcfb31ec300596d7f44d3348) 
2024-05-24T20:17:09.6674929Z     ==44942==ABORTING
```

audrow, 2024-06-06:
FYI @MiguelCompany and @EduPonz

fujitatomoya, 2024-06-06:
this seems that https://github.com/ros2/rmw_fastrtps/issues/761 could be related but this one happens with iron and humble, so can be another issue.

CC: @Barry-Xu-2018

Barry-Xu-2018, 2024-06-07:
@luca-della-vedova  

I try to reproduce this problem on my environment (docker image: osrf/ros:iron-desktop-full-jammy).  
But I cannot get the same error log as yours.

If run `colcon test --packages-up-to rmf_fleet_adapter`, “1: AddressSanitizer:DEADLYSIGNAL” keeps being output while executing test_rmf_utils
```
    Start 1: test_rmf_utils

1: Test command: /usr/bin/python3.10 "-u" "/opt/ros/iron/share/ament_cmake_test/cmake/run_test.py" "/root/iron_rmf_ws/build/rmf_utils/test_results/rmf_utils/test_rmf_utils.catch2.xml" "--package-name" "rmf_utils" "--output-file" "/root/iron_rmf_ws/build/rmf_utils/ament_cmake_catch2/test_rmf_utils.txt" "--command" "/root/iron_rmf_ws/build/rmf_utils/test_rmf_utils" "-r junit" "-o /root/iron_rmf_ws/build/rmf_utils/test_results/rmf_utils/test_rmf_utils.catch2.xml"
1: Test timeout computed to be: 300
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_internal_msgs/rmf_site_map_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_site_map_msgs/test_results/rmf_site_map_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_site_map_msgs/rosidl_generator_c/rmf_site_map_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_internal_msgs/rmf_fleet_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_fleet_msgs/test_results/rmf_fleet_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_fleet_msgs/rosidl_generator_c/rmf_fleet_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_internal_msgs/rmf_dispenser_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_dispenser_msgs/test_results/rmf_dispenser_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_dispenser_msgs/rosidl_generator_c/rmf_dispenser_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_internal_msgs/rmf_door_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_door_msgs/test_results/rmf_door_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_door_msgs/rosidl_generator_c/rmf_door_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_internal_msgs/rmf_traffic_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_traffic_msgs/test_results/rmf_traffic_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_traffic_msgs/rosidl_generator_c/rmf_traffic_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_building_map_msgs/rmf_building_map_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_building_map_msgs/test_results/rmf_building_map_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_building_map_msgs/rosidl_generator_c/rmf_building_map_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/src/rmf/rmf_internal_msgs/rmf_lift_msgs':
1:  - /opt/ros/iron/bin/ament_cppcheck --xunit-file /root/iron_rmf_ws/build/rmf_lift_msgs/test_results/rmf_lift_msgs/cppcheck_rosidl_generated_c.xunit.xml /root/iron_rmf_ws/build/rmf_lift_msgs/rosidl_generator_c/rmf_lift_msgs
1: -- run_test.py: invoking following command in '/root/iron_rmf_ws/build/rmf_utils':
1:  - /root/iron_rmf_ws/build/rmf_utils/test_rmf_utils -r junit -o /root/iron_rmf_ws/build/rmf_utils/test_results/rmf_utils/test_rmf_utils.catch2.xml
1: AddressSanitizer:DEADLYSIGNAL
1: AddressSanitizer:DEADLYSIGNAL
1: AddressSanitizer:DEADLYSIGNAL
1: AddressSanitizer:DEADLYSIGNAL
1: AddressSanitizer:DEADLYSIGNAL
1: AddressSanitizer:DEADLYSIGNAL
...
```

If run `colcon test --packages-select rmf_fleet_adapter`, I get below output
```
    Start 2: test_rmf_fleet_adapter

2: Test command: /usr/bin/python3.10 "-u" "/opt/ros/iron/share/ament_cmake_test/cmake/run_test.py" "/root/iron_rmf_ws/build/rmf_fleet_adapter/test_results/rmf_fleet_adapter/test_rmf_fleet_adapter.catch2.xml" "--package-name" "rmf_fleet_adapter" "--output-file" "/root/iron_rmf_ws/build/rmf_fleet_adapter/ament_cmake_catch2/test_rmf_fleet_adapter.txt" "--command" "/root/iron_rmf_ws/build/rmf_fleet_adapter/test_rmf_fleet_adapter" "-r junit" "-o /root/iron_rmf_ws/build/rmf_fleet_adapter/test_results/rmf_fleet_adapter/test_rmf_fleet_adapter.catch2.xml"
2: Test timeout computed to be: 300
2: -- run_test.py: invoking following command in '/root/iron_rmf_ws/build/rmf_fleet_adapter':
2:  - /root/iron_rmf_ws/build/rmf_fleet_adapter/test_rmf_fleet_adapter -r junit -o /root/iron_rmf_ws/build/rmf_fleet_adapter/test_results/rmf_fleet_adapter/test_rmf_fleet_adapter.catch2.xml
2: =================================================================
2: ==43572==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x606000120e00 in thread T0:
2:   object passed to delete has wrong type:
2:   size of the allocated type:   64 bytes;
2:   size of the deallocated type: 1 bytes.
2:     #0 0x706b7ff3424f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172
2:     #1 0x59d0f715991a in __gnu_cxx::new_allocator<char>::deallocate(char*, unsigned long) (/root/iron_rmf_ws/build/rmf_fleet_adapter/test_rmf_fleet_adapter+0x138d91a)
[... cut ...]
2:     #3 0x59d0f737bd4b in void* rclcpp::allocator::retyped_allocate<std::allocator<char> >(unsigned long, void*) (/root/iron_rmf_ws/build/rmf_fleet_adapter/test_rmf_fleet_adapter+0x15afd4b)
2:     #4 0x706b7459f3d2 in rcutils_string_map_init (/opt/ros/iron/lib/librcutils.so+0xe3d2)
2: 
2: SUMMARY: AddressSanitizer: new-delete-type-mismatch ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172 in operator delete(void*, unsigned long)
2: ==43572==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
2: ==43572==ABORTING

```

Could you provide your complete log?

Barry-Xu-2018, 2024-06-07:
> Indeed there are a lot of new-delete-type-mismatch errors but I believe those have always been around.

Yes. This is a known issue. (https://github.com/ros2/rclcpp/issues/2220)

> We do set in our CI the [ASAN_OPTIONS](https://github.com/open-rmf/rmf_ci_templates/blob/634304014024ff4e2bf9ee6c2d81e7b8ddefc54a/.github/workflows/reusable_build.yaml#L81) environment variable to avoid these spurious failures

Thanks. I get the same error. 
```
2: [INFO] [1717746840.951931995] [test_Delivery]: Executing go_to_place [dropoff] for robot [test_fleet/T0]
2: cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'
2: Fail in delete datareader, at ./src/rmw_service.cpp:100 during '__function__'
2: AddressSanitizer:DEADLYSIGNAL
2: =================================================================
2: ==185==ERROR: AddressSanitizer: SEGV on unknown address 0x0000000007f0 (pc 0x7c812d256ef4 bp 0x60c0006d9a00 sp 0x7ffc3f3cc5e8 T0)
2: ==185==The signal is caused by a READ memory access.
2: ==185==Hint: address points to the zero page.
2:     #0 0x7c812d256ef4 in pthread_mutex_lock (/lib/x86_64-linux-gnu/libc.so.6+0x97ef4)
2:     #1 0x7c812772f20d in eprosima::fastdds::dds::DomainParticipantImpl::find_type(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const (/opt/ros/iron/lib/libfastrtps.so.2.10+0x34820d)
2:     #2 0x7c8127731735 in eprosima::fastdds::dds::DomainParticipantImpl::unregister_type(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/opt/ros/iron/lib/libfastrtps.so.2.10+0x34a735)
2:     #3 0x7c8127ce28ed in rmw_fastrtps_shared_cpp::__rmw_destroy_service(char const*, rmw_node_s*, rmw_service_s*) (/opt/ros/iron/lib/librmw_fastrtps_shared_cpp.so+0x4e8ed)
2:     #4 0x7c81309e889b in rcl_service_fini (/opt/ros/iron/lib/librcl.so+0x2b89b)
2:     #5 0x7c81309e8c52 in rcl_node_type_description_service_fini (/opt/ros/iron/lib/librcl.so+0x2bc52)
2:     #6 0x7c8130c461f3  (/opt/ros/iron/lib/librclcpp.so+0x14c1f3)
2:     #7 0x7c8130c274e7  (/opt/ros/iron/lib/librclcpp.so+0x12d4e7)
2:     #8 0x7c8130c25fe6  (/opt/ros/iron/lib/librclcpp.so+0x12bfe6)
2:     #9 0x7c812d204a55 in __cxa_finalize (/lib/x86_64-linux-gnu/libc.so.6+0x45a55)
2:     #10 0x7c8130be84d6  (/opt/ros/iron/lib/librclcpp.so+0xee4d6)
2: 
2: AddressSanitizer can not provide additional info.
2: SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libc.so.6+0x97ef4) in pthread_mutex_lock
2: ==185==ABORTING
```

MiguelCompany, 2024-06-07:
@luca-della-vedova @Barry-Xu-2018 since this is inside `__cxa_finalize `, I guess it has to do with the destruction order of static global objects. So I agree with https://github.com/ros2/rmw_fastrtps/issues/760#issuecomment-2153124168 that this is (in a way) similar to #761

Barry-Xu-2018, 2024-06-07:
@MiguelCompany  

For #761, I found a bug in the Executor. It causes shared pointers to resources not to be released in time. Considering that the Jazzy version of the Executor has significant changes, I am uncertain whether the Iron version of the Executor has a similar issue.

Barry-Xu-2018, 2024-06-12:
The issue is related to test `Test Delivery` in rmf_fleet_adapter/test/tasks/test_Delivery.cpp.  

For `cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'`, backstrace output 

```c++
#0  rmw_fastrtps_shared_cpp::__rmw_publish (identifier=0x7fffe8c86192 "rmw_fastrtps_cpp", publisher=0x555556095420, ros_message=0x7fffffff5140, allocation=0x0)
    at /root/ros2_iron_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:62
#1  0x00007fffe8b953c6 in rmw_fastrtps_shared_cpp::__rmw_destroy_service (identifier=0x7fffe8c86192 "rmw_fastrtps_cpp", node=0x5555560aa5d0, service=0x555556211690)
    at /root/ros2_iron_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_service.cpp:76
#2  0x00007fffe8c6e8d0 in rmw_destroy_service (node=0x5555560aa5d0, service=0x555556211690) at /root/ros2_iron_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_service.cpp:506
#3  0x00007fffeb44dc1f in rmw_destroy_service (v2=0x5555560aa5d0, v1=0x555556211690) at /root/ros2_iron_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:523
#4  0x00007fffeeda0996 in rcl_service_fini (service=0x555555f452e8, node=0x555555f45200) at /root/ros2_iron_ws/src/ros2/rcl/rcl/src/rcl/service.c:260
#5  0x00007fffeed99af3 in rcl_node_type_description_service_fini (node=0x555555f45200) at /root/ros2_iron_ws/src/ros2/rcl/rcl/src/rcl/node.c:673
#6  0x00007fffefa15127 in rclcpp::node_interfaces::NodeTypeDescriptions::NodeTypeDescriptionsImpl::~NodeTypeDescriptionsImpl (this=0x5555561f6630, __in_chrg=<optimized out>)
    at /root/ros2_iron_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_type_descriptions.cpp:132
#7  0x00007fffefa159f4 in std::default_delete<rclcpp::node_interfaces::NodeTypeDescriptions::NodeTypeDescriptionsImpl>::operator() (this=0x5555561dda18, __ptr=0x5555561f6630)
    at /usr/include/c++/11/bits/unique_ptr.h:85
#8  0x00007fffefa156f0 in std::unique_ptr<rclcpp::node_interfaces::NodeTypeDescriptions::NodeTypeDescriptionsImpl, std::default_delete<rclcpp::node_interfaces::NodeTypeDescriptions::NodeTypeDescriptionsImpl> >::~unique_ptr (this=0x5555561dda18, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/unique_ptr.h:361
#9  0x00007fffefa10d12 in rclcpp::node_interfaces::NodeTypeDescriptions::~NodeTypeDescriptions (this=0x5555561dda10, __in_chrg=<optimized out>)
    at /root/ros2_iron_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_type_descriptions.cpp:154
#10 0x00007fffef9b108f in __gnu_cxx::new_allocator<rclcpp::node_interfaces::NodeTypeDescriptions>::destroy<rclcpp::node_interfaces::NodeTypeDescriptions> (this=0x5555561dda10, __p=0x5555561dda10)
    at /usr/include/c++/11/ext/new_allocator.h:168
#11 0x00007fffef9b1005 in std::allocator_traits<std::allocator<rclcpp::node_interfaces::NodeTypeDescriptions> >::destroy<rclcpp::node_interfaces::NodeTypeDescriptions> (__a=..., __p=0x5555561dda10)
    at /usr/include/c++/11/bits/alloc_traits.h:535
#12 0x00007fffef9b09a5 in std::_Sp_counted_ptr_inplace<rclcpp::node_interfaces::NodeTypeDescriptions, std::allocator<rclcpp::node_interfaces::NodeTypeDescriptions>, (__gnu_cxx::_Lock_policy)2>::_M_dispose
    (this=0x5555561dda00) at /usr/include/c++/11/bits/shared_ptr_base.h:528
#13 0x0000555555ad9495 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5555561dda00) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#14 0x0000555555ac8d7d in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x5555561f6538, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:705
#15 0x00007fffef9a5848 in std::__shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x5555561f6530, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#16 0x00007fffef9a5892 in std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface>::~shared_ptr (this=0x5555561f6530, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr.h:122
#17 0x00007fffef9aee54 in std::pair<rclcpp::Node const* const, std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface> >::~pair (this=0x5555561f6528, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/stl_pair.h:211
#18 0x00007fffef9aee78 in __gnu_cxx::new_allocator<std::__detail::_Hash_node<std::pair<rclcpp::Node const* const, std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface> >, false> >::destroy<std::pair<rclcpp::Node const* const, std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface> > > (this=0x7fffefdab100 <rclcpp::Node::backport_members_>, __p=0x5555561f6528)
    at /usr/include/c++/11/ext/new_allocator.h:168
#19 0x00007fffef9adcaf in std::allocator_traits<std::allocator<std::__detail::_Hash_node<std::pair<rclcpp::Node const* const, std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface> >, false> > >::destroy<std::pair<rclcpp::Node const* const, std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface> > > (__a=..., __p=0x5555561f6528) at /usr/include/c++/11/bits/alloc_traits.h:535
#20 0x00007fffef9ac7fb in std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<rclcpp::Node const* const, std::shared_ptr<rclcpp::node_interfaces::NodeTypeDescriptionsInterface> >, false> > >::_M_deallocate_node (this=0x7fffefdab100 <rclcpp::Node::backport_members_>, __n=0x5555561f6520) at /usr/include/c++/11/bits/hashtable_policy.h:1894
[... cut ...]
Node::BackportMembers is a static variable in rclcpp::Node.  So it is destructed at __cxa_finalize. 
https://github.com/ros2/rclcpp/blob/c1a01fc08d728bcf06f3632939b1b6c06f1b99a2/rclcpp/include/rclcpp/node.hpp#L1604

When the Node is destructed, it will call `backport_members_.remove(this)`.  
The current issue is why the Node `test_Delivery` wasn't released.   
Further investigation requires understanding the relevant test code first. However, I'm not familiar with the rmf_fleet_adapter code. Can anyone provide assistance?

Barry-Xu-2018, 2024-06-13:
@MiguelCompany  

Thanks.

> Could you check if https://github.com/ros2/rmw_fastrtps/pull/770 fixes this?

After checking, the below error message doesn't occur. 
```
2: cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'
2: Fail in delete datareader, at ./src/rmw_service.cpp:100 during '__function__'
```

For test `Test Delivery`, the destructor function of Node `test_Delivery` is also never called. After calling destructor function of Node::BackportMembers (Note that it is a static variable in rclcpp::Node.), the resource (service, publisher, etc) of Node `test_Delivery` are released.  So I think there is another issue.

fujitatomoya, 2024-06-15:
@Barry-Xu-2018 

I think this issue is originally generated by https://github.com/ros2/rclcpp/pull/2224.

maybe https://github.com/ros2/rcl/pull/1112, https://github.com/ros2/rclcpp/pull/2344 and https://github.com/ros2/rclcpp/pull/2351 could address the issue, but i am not 100% sure.

since https://github.com/ros2/rcl/pull/1112 requires API change, backport is not straight forward.

@MiguelCompany i will try to take a look.

Barry-Xu-2018, 2024-06-15:
@fujitatomoya 

> I think this issue is originally generated by https://github.com/ros2/rclcpp/pull/2224.

That could be it. The error occurred while releasing service /test_Delivery/get_type_description.

volleypaul, 2024-10-10:
Is this still an issue in Jazzy?  Thanks.

fujitatomoya, 2024-10-10:
@volleypaul i have not tried with jazzy but https://github.com/ros2/rmw_fastrtps/pull/770 is not merged yet. so i am almost sure the problem still stays.

riv-ljames, 2025-11-21:
I believe I am encountering this issue in Jazzy.