ros2/rmw_fastrtps #774: Fastrtps udp packets are larger than CycloneDDS packets

rty813, 2024-08-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu
- Installation type:
  - from source
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS and CycloneDDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Start the publisher on one host and the subscriber on another, then capture packets using tcpdump. 
publisher: `ros2 topic pub -r 100 test std_msgs/String "{data: asdf}"`
subscriber: `ros2 topic hz test`

#### Expected behavior
The packet sizes are about the same.

#### Actual behavior
FastRTPS packets are about 50 bytes larger than CycloneDDS packets.
FastRTPS:
[image]

CycloneDDS:
[image]

rty813, 2024-08-24:
[pcap.zip](https://github.com/user-attachments/files/16734922/pcap.zip)

MiguelCompany, 2024-08-27:
@rty813 

In order to help you better, we'd need more information here.

You say that you are building from sources, but you are not indicating which version you are building (rolling? jazzy?)
It would be best if you point to the commit hash of the main repo (i.e. https://github.com/ros2/ros2).
You should also indicate whether you are using any build flags different from the ones indicated in [the documentation](https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html).

I tried reproducing with the Jazzy docker image, and I only see `INFO_TS` and `DATA` sub-messages, but no HEARTBEATs, since the default configuration is `BEST_EFFORT`. So I suppose you either built the stack with different default values, or you are providing some custom configuration via environment variables or something.

rty813, 2024-08-28:
@MiguelCompany The compiled version is Foxy, which is no longer maintained. 

I tried it in the humble docker image and found that the data volume of fastrtps is still a little higher than that of cyclonedds.

test command: `ros2 topic pub -r100 testt std_msgs/String "{data: hellowaslkdjflaksdjfklasdforld}"`

cyclonedds:
[image]

fastrtps:
[image]

MiguelCompany, 2024-09-19:
@rty813 Again, please provide full instructions on how to reproduce, including the commands to measure the bandwidth and, if possible, packet captures that we can analyze.

rty813, 2024-09-20:
First, I ran a ROS Docker image on two separate hosts to avoid using shm. Then, I executed `topic pub` on one host and `topic echo` on the other. I used the `ifstat 1` command to periodically print the network interface rate.

MiguelCompany, 2024-10-09:
@rty813 Some things that could be considered:

* There are several timing configurations that could affect the generated bandwidth. In particular, [heartbeat period](https://fast-dds.docs.eprosima.com/en/stable/fastdds/use_cases/large_data/large_data.html#tuning-heartbeat-period) and [participant announcement period](https://fast-dds.docs.eprosima.com/en/stable/fastdds/discovery/general_disc_settings.html#announcement-period) come to mind.
* Fast DDS adds INFO_DST submessage when the packets are sent to a single participant.
* Fast DDS add custom statistics submessage. This can be avoided disabling the statistics when building Fast DDS by setting `-DFASTDDS_STATISTICS=OFF` [cmake option](https://fast-dds.docs.eprosima.com/en/stable/installation/configuration/cmake_options.html#general-options)

MiguelCompany, 2024-10-15:
Closing since enough information was provided

closed 2024-10-15 by MiguelCompany