ros2/rmw_fastrtps #775: Service request string type fails with microROS

hr-ti, 2024-08-26:
## Bug report

**Required Info:**
I'm using microROS for my STM32 board and attempting to use services. The issue I'm getting on the MCU side is that deserializing the string type of a service request results in an error where the string length exceeds the capacity of the buffer, which is defaulting to 0 (erroring [here](https://github.com/eProsima/Micro-CDR/blob/master/src/c/types/sequence.c#L33)). 

I've verified that UDP traffic is working correctly. So I can only suspect that the host side is incorrectly converting service requests either in the RMW or the agent, not too sure how to confirm so input here would be great.

For my env, I use `micro-ros-agent` to handle bridging traffic between DDSes, but since the agent is built on top of FastDDS, service interoperability [does not work ](https://github.com/ros2/rmw_cyclonedds/issues/184) directly. To workaround this, I send the raw data through a socket that pipes to a FastDDS node that communicates to the agent exclusively. This works perfectly for service requests that **don't** involve a string type in its request. 

A good note to mention is that I've tried this in a `rmw_fastrtps_cpp` environment and the error persists, so it's not an interoperability issue. 

Running my firmware through a debugger shows that it casts the appropriate service request typesupport onto the incoming ros_message once a service request is found, but capacity of the string type is always 0. Here's the generated typesupport code:

```
static bool _SetString_Request__cdr_deserialize(
  ucdrBuffer * cdr,
  void * untyped_ros_message)
{
  (void) cdr;

  bool rv = false;

  if (!untyped_ros_message) {
    return false;
  }
  _SetString_Request__ros_msg_type * ros_message = (_SetString_Request__ros_msg_type *)(untyped_ros_message);
  (void)ros_message;

  // Field name: data
  {
    size_t capacity = ros_message->data.capacity;
    uint32_t string_size;
    rv = ucdr_deserialize_sequence_char(cdr, ros_message->data.data, capacity, &string_size);
    if (rv) {
      ros_message->data.size = (string_size == 0) ? 0 : string_size - 1;
    } else if(string_size > capacity){
      cdr->error = false;
      cdr->last_data_size = 1;
      ros_message->data.size = 0;
      ucdr_align_to(cdr, sizeof(char));
      ucdr_advance_buffer(cdr, string_size);
    }
  }
  return rv;
```
My understanding could be wrong, but `udr_deserialize_sequence_char` uses `ros_message->data.capacity` to copy the appropriate amount of the input buffer `cdr` to `ros_message`. Here, `ros_message->data.capacity` is 0, `udr_deserialize_sequence_char` fails, `rv` is false and therefore causes an executor error. 

Viewing the cdr buffer, I can see that the string I pass through is correct, including its length (+ null terminator), but `ros_message` attributes are incorrect.

I'm lost on where the buffer capacity property is set, but I'm hoping this is a user error and not an error from RMW.

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binary
- Version or commit hash:
  - humble
- DDS implementation:
  - FastDDS -> MicroXRCE-DDS
- Client library (if applicable):
  - rclpy (host), rclc (MCU)

#### Steps to reproduce issue
1. Make a service server on the MCU side, takes in a service request that has a string type
2. Start `micro-ros-agent` and connect to MCU
3. Call service from host

#### Expected behavior
The service server properly reads the request and handles appropriately

#### Actual behavior
Executor fails when deserializing the request

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hr-ti, 2024-08-26:
After some testing, I realize this error persists when using a sequence or array of any type. Since capacity is always 0, only empty arrays work

MiguelCompany, 2024-08-27:
@hr-ti I was going to tell you it was better to report this in any of the Micro-ROS repositories.

I see you already opened https://github.com/eProsima/Micro-XRCE-DDS/issues/178, which has already been answered, so I am closing this one as duplicate.

closed 2024-08-27 by MiguelCompany