ros2/rmw_fastrtps #786: ROS 2 Humble node fails to register in Docker with host when both wifi and ethernet are active

pierrot32, 2024-10-24:
## Bug report

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS2 Humble
- DDS implementation:
  - Tried rmw_fastrtps_cpp with an interface whitelist on the Ethernet interface.
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1. Set up ROS 2 Humble in a Docker container using the ros:humble-ros-base image.
2. Implement a basic "HelloWorld" publisher node.
3. Set the Docker container's network_mode and pid to host to allow communication between the host system and another machine on the same network.
4. Launch the node in the Docker container.
5. Attempt to subscribe to the topic from the host machine using tools like rqt or ros2 topic echo /topic.
6. Test with:
  - WiFi active only: Everything works as expected.
  - Ethernet active only: Everything works as expected.
  - Both WiFi and Ethernet active simultaneously: The node fails to register, and issues occur.
 
### Expected behavior
The ROS 2 node should register correctly, and I should be able to subscribe to the topic using ros2 topic echo /topic or other ROS 2 tools, regardless of whether both WiFi and Ethernet are active.

### Actual Behavior:
- When both WiFi and Ethernet are active:
   - The topic is visible using ros2 topic echo /topic from the host machine.
   - However, the node does not register properly, and ros2 topic info /topic -v shows:
     - Node name: _NODE_NAME_UNKNOWN_
     - Node namespace: _NODE_NAMESPACE_UNKNOWN_

### Additional Information:
- I attempted to configure the rmw_fastrtps_cpp implementation with a whitelist specifying only the Ethernet interface, but this did not resolve the issue.
- When using network_mode and pid not set to host, everything works correctly, even with both WiFi and Ethernet active.
- There are no errors displayed in the terminal.

christophebedard, 2024-10-24:
I moved this to the https://github.com/ros2/rmw_fastrtps repo for now. It's nice if it works with Cyclone DDS, but there still might be an issue with Fast DDS.

> The node fails to register

Can you explain what you mean by this?

Looks like `ros2 topic info /topic -v` simply outputs `Unknown topic '/topic'` if `/topic` doesn't exist, so it's weird that you're not getting that.

Do both interfaces exist before you run anything, and do any of them get disabled some time later? If one of them is disabled in between running your node and `ros2 topic` (e.g., because of a hardware issue with your dock), it might affect it. Something like https://discourse.ros.org/t/network-issues-ros-topic-visibility-dependency-to-network-interfaces/29385. However, you're explicitly asking Fast DDS to only use Ethernet (can you provide more information on how you're doing that?).

I can't personally test this right now (no ethernet), but I can give it a try probably next week.

pierrot32, 2024-10-24:
When I said 'the node fails to register', I meant that the node doesn’t appear when using the ros2 node list command, making it inaccessible. However, this behavior is inconsistent. For example, right now, the node is visible in the list, but when I run ros2 topic echo /topic, no data is being published to the topic, even though data is expected.

Yes, both interfaces (WiFi and Ethernet) are active before I run the Docker container, and no, I’m not stopping any interface during execution.

To apply a whitelist to the Ethernet interface, I created a custom XML file based on the documentation at [Fast DDS Transport Whitelist](https://fast-dds.docs.eprosima.com/en/v3.1.0/fastdds/transport/whitelist.html). However, I’m not entirely sure if the XML is configured correctly. I tried to follow the documentation, but I’m uncertain if it’s working as expected.

Here’s the XML I used:
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com">
   <transport_descriptors>
        <transport_descriptor>
            <transport_id>CustomTcpTransportWhitelistAddress</transport_id>
            <type>TCPv4</type>
            <interfaceWhiteList>
                <address>enx806d974d0280</address> <!-- Removed the colon -->
            </interfaceWhiteList>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="participant_profile_ros2" is_default_profile="true">
        <rtps>
            <name>profile_for_ros2_context</name>
        </rtps>
    </participant>
</profiles>
```
Since everything is working with Cyclone DDS, I’m not in a rush to solve this issue with Fast DDS. Please take your time, and if you think of any possible solutions, I’d be happy to try them out on my system!

MiguelCompany, 2024-10-25:
@pierrot32 @christophebedard 

I think this is a manifestation of https://github.com/eProsima/Fast-DDS/discussions/5104 plus Fast DDS using a hash on the IP addresses of the network interfaces to decide whether to use the shared memory transport

Regarding the XML, it has several issues.
One of them is trying to use an interface name for the whitelist, which is only supported in Jazzy.
Another one is that you should disable the builtin transports, and add the ones you want to customize.
An example of this is [here](https://fast-dds.docs.eprosima.com/en/2.6.x/fastdds/transport/udp/udp.html#enabling-udp-transport)

I would suggest the following XML, which forces UDP only communications through the interface with the specified address:

```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com">
   <transport_descriptors>
        <transport_descriptor>
            <transport_id>CustomUDPTransportWithWhitelist</transport_id>
            <type>UDPv4</type>
            <interfaceWhiteList>
                <address>192.168.2.2</address> <!-- Change with the address of your ethernet device -->
            </interfaceWhiteList>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="participant_profile_ros2" is_default_profile="true">
        <rtps>
            <useBuiltinTransports>false</useBuiltinTransports>
            <userTransports>
                <transport_id>CustomUDPTransportWithWhitelist</transport_id>
            </userTransports>
        </rtps>
    </participant>
</profiles>
```

MiguelCompany, 2024-10-27:
@pierrot32 I suspect Cyclone DDS might use bigger socket buffer sizes by default, and also make use of non-blocking send calls.

The following is an updated version of the previous XML:

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com">
   <transport_descriptors>
        <transport_descriptor>
            <transport_id>CustomUDPTransportWithWhitelist</transport_id>
            <type>UDPv4</type>

            <!-- 
                Use 1 million bytes on socket buffer sizes.
                A good value here would be 1.5 times the highest burst of data sent by the application.
            -->
            <sendBufferSize>1000000</sendBufferSize>
            <receiveBufferSize>1000000</receiveBufferSize>

            <!--
                The following makes the send calls non-blocking.
                This avoids the publisher blocking until there is room
                in the socket buffer for the datagram being sent, but then
                increases the datagrams lost when there is not enough buffer.
            -->
            <non_blocking_send>true</non_blocking_send>

            <interfaceWhiteList>
                <address>192.168.2.2</address> <!-- Change with the address of your ethernet device -->
            </interfaceWhiteList>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="participant_profile_ros2" is_default_profile="true">
        <rtps>
            <useBuiltinTransports>false</useBuiltinTransports>
            <userTransports>
                <transport_id>CustomUDPTransportWithWhitelist</transport_id>
            </userTransports>
        </rtps>
    </participant>
</profiles>
```

pierrot32, 2024-10-29:
Changing the buffersize, and make use of non-blocking send calls works. 

Thank you @christophebedard and @MiguelCompany for helping me!

closed 2024-10-29 by pierrot32