ros2/rmw_fastrtps #789: SHM transport: The longer it runs, The more latency increases. (Buffer Full Exception)

fujitatomoya, 2024-11-20:
### Description

`tf` / `tf_static` topics start generating the dropping message warning, 'the timestamp on the message is earlier than all the data in the transform cache`.
once it starts happening, the frequency just increases linearly as time goes.
this actually increases the delay and latency between publisher and subscribers significantly.

### System Information

- ros2: Humble, patch release 5
- ros-humble-fastrtps: v2.6.7 with + https://github.com/eProsima/Fast-DDS/pull/4708
- Number of Nodes: - 120
- Number of Topics: - 500
- `tf` / `tf_static` topic: Publisher -15 and Subscriber 30 for each topic.

### XML configuration

```xml
<?xml version="1.0" encoding="UTF-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <profiles>
        <transport_descriptors>
            <!-- Create a descriptor for the new transport -->
            <transport_descriptor>
                <transport_id>shm_transport</transport_id>
                <type>SHM</type>
                <segment_size>10485760</segment_size> 
                <port_queue_capacity>256</port_queue_capacity>
            </transport_descriptor>
        </transport_descriptors>
        <participant profile_name="SHMParticipant" is_default_profile="true">
            <rtps>
                <!-- Link the Transport Layer to the Participant -->
                <userTransports>
                  <transport_id>shm_transport</transport_id>
                </userTransports>
                <!-- <useBuiltinTransports>false</useBuiltinTransports> -->
            </rtps>
        </participant>
        <publisher profile_name="service">
            <qos>
                <reliability>
                    <max_blocking_time>
                        <sec>10</sec>
                    </max_blocking_time>
                </reliability>
            </qos>
        </publisher>
    </profiles>
</dds>
```

### Additional Information

- When this warning happens, https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp#L228-L231 this if statement comes True.
- if we increase `port_queue_capacity` increases 4 times bigger, the 4 times longer it takes until the problem starts. (almost the same situation for `segment_size`) this does not avoid this original problem, it only can buy some time based on the configuration.

fujitatomoya, 2024-11-20:
@MiguelCompany we have detected this issue with production environment. we will give it a shot if falling back to `udp_transport` see if the problem happens. do you happen know any known issue for this?

Barry-Xu-2018, 2024-11-20:
I have some questions about the code in `class MultiProducerConsumerRingBuffer`. If my understanding is incorrect, please correct me.

Assuming there's currently only 1 free cell (that is, ``free_cells`` is 1)  

At this moment, a data will push to a cell. It will throw an exception "buffer full". At this time, the value of `free_cells` is changed from 1 to 0.  So, even though no cell was used, the value of the `free_cells`  still decreased by 1.

https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp#L219-L232
```c++
    bool push(
            const T& data)
    {
        // If no listeners the buffer is dropped
        if (node_->registered_listeners_ == 0)
        {
            return false;
        }

        auto pointer = node_->pointer_.load(std::memory_order_relaxed);

        // if free cells, increase the write pointer and decrease the free cells
        while (pointer.ptr.free_cells > 0 &&
                !node_->pointer_.compare_exchange_weak(pointer,
                { { inc_pointer(pointer.ptr.write_p), pointer.ptr.free_cells - 1 } },
                std::memory_order_release,
                std::memory_order_relaxed))
        {
        }

        if (pointer.ptr.free_cells == 0)
        {
            throw std::runtime_error("Buffer full");
        }
  ...
 }
```

If a listener pop cell, and it is the last listener on this cell, this cell can be reused (`free_cells` should be increased).  
At this time, `free_cells` become 1. 
https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp#L120-L131
```c++
        bool pop()
        {
            ...

            // If all the listeners have read the cell
            if (counter == 1)
            {
                // Increase the free cells => increase the global read pointer
                auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);
                while (!buffer_.node_->pointer_.compare_exchange_weak(pointer,
                        { { pointer.ptr.write_p, pointer.ptr.free_cells + 1 } },
                        std::memory_order_release,
                        std::memory_order_relaxed))
                {
                }
            }
````

And then, a new data will push to a cell. Then we'll face the same issue --- the push didn't succeed, but the `free_cells` are reduced to 0 again.   
This will cause the available cells not to be used and will always be in a buffer full state.

MiguelCompany, 2024-11-20:
> At this moment, a data will push to a cell. It will throw an exception "buffer full".

The exception would never be thrown if the current thread successfully updated the counter from 1 to 0.
In that case, the local value of `pointer.ptr.free_cells` would be `1`.

You might want to try upgrading all `node_->pointer_.load()` calls to `std::memory_order_acquire`, and check whether the situation improves

fujitatomoya, 2024-11-20:
@Barry-Xu-2018 you are saying, this if statement should be moved above?

```patch
diff --git a/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp b/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp
index 51e559155..52e2be7f4 100644
--- a/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp
+++ b/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp
@@ -216,6 +216,11 @@ public:

         auto pointer = node_->pointer_.load(std::memory_order_relaxed);

+        if (pointer.ptr.free_cells == 0)
+        {
+            throw std::runtime_error("Buffer full");
+        }
+
         // if free cells, increase the write pointer and decrease the free cells
         while (pointer.ptr.free_cells > 0 &&
                 !node_->pointer_.compare_exchange_weak(pointer,
@@ -225,11 +230,6 @@ public:
         {
         }

-        if (pointer.ptr.free_cells == 0)
-        {
-            throw std::runtime_error("Buffer full");
-        }
-
         auto& cell = cells_[get_pointer_value(pointer.ptr.write_p)];

         cell.data(data);
```

fujitatomoya, 2024-11-20:
> You might want to try upgrading all node_->pointer_.load() calls to std::memory_order_acquire, and check whether the situation improves

okay that guarantees that previous write operation with other threads before `release` to be visible when it loads the `std::atomic<PtrType>`. the expectation is that other threads are writing the data to this atomic field but it is not synchronized yet. and that could lead the buffer full situation with many threads are running?

```patch
diff --git a/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp b/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp
index 51e559155..d2be85ada 100644
--- a/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp
+++ b/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp
@@ -85,7 +85,7 @@ public:
          */
         Cell* head()
         {
-            auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);
+            auto pointer = buffer_.node_->pointer_.load(std::memory_order_acquire);

             // If local read_pointer and write_pointer are equal => buffer is empty for this listener
             if (read_p_ == pointer.ptr.write_p )
@@ -121,7 +121,7 @@ public:
             if (counter == 1)
             {
                 // Increase the free cells => increase the global read pointer
-                auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);
+                auto pointer = buffer_.node_->pointer_.load(std::memory_order_acquire);
                 while (!buffer_.node_->pointer_.compare_exchange_weak(pointer,
                         { { pointer.ptr.write_p, pointer.ptr.free_cells + 1 } },
                         std::memory_order_release,
@@ -214,7 +214,7 @@ public:
             return false;
         }

-        auto pointer = node_->pointer_.load(std::memory_order_relaxed);
+        auto pointer = node_->pointer_.load(std::memory_order_acquire);

         // if free cells, increase the write pointer and decrease the free cells
         while (pointer.ptr.free_cells > 0 &&
@@ -240,12 +240,12 @@ public:

     bool is_buffer_full()
     {
-        return (node_->pointer_.load(std::memory_order_relaxed).ptr.free_cells == 0);
+        return (node_->pointer_.load(std::memory_order_acquire).ptr.free_cells == 0);
     }

     bool is_buffer_empty()
     {
-        return (node_->pointer_.load(std::memory_order_relaxed).ptr.free_cells == node_->total_cells_);
+        return (node_->pointer_.load(std::memory_order_acquire).ptr.free_cells == node_->total_cells_);
     }

     /**
@@ -261,7 +261,7 @@ public:
         // The new listener's read pointer is the current write pointer
         auto listener = std::unique_ptr<Listener>(
             new Listener(
-                *this, node_->pointer_.load(std::memory_order_relaxed).ptr.write_p));
+                *this, node_->pointer_.load(std::memory_order_acquire).ptr.write_p));

         node_->registered_listeners_++;

@@ -293,7 +293,7 @@ public:
     {
         if (node_->registered_listeners_ > 0)
         {
-            auto pointer = node_->pointer_.load(std::memory_order_relaxed);
+            auto pointer = node_->pointer_.load(std::memory_order_acquire);

             uint32_t p = pointer_to_head(pointer);

```

Barry-Xu-2018, 2024-11-21:
@MiguelCompany 

> The exception would never be thrown if the current thread successfully updated the counter from 1 to 0.
In that case, the local value of pointer.ptr.free_cells would be 1.

Thanks. I misunderstand.  

> You might want to try upgrading all node_->pointer_.load() calls to std::memory_order_acquire, and check whether the situation improves

```c++
        // if free cells, increase the write pointer and decrease the free cells
        while (pointer.ptr.free_cells > 0 &&
                !node_->pointer_.compare_exchange_weak(pointer,
                { { inc_pointer(pointer.ptr.write_p), pointer.ptr.free_cells - 1 } },
                std::memory_order_release,
                std::memory_order_relaxed))
        {
        }

        if (pointer.ptr.free_cells == 0)
        {
            throw std::runtime_error("Buffer full");
        }
```

No. I think std::memory_order_relaxed is unsuitable, while compare_exchange_weak is failure.   
While compare_exchange_weak return failure, `inode_->pointer_`  will be loaded to pointer. And following comparison "pointer.ptr.free_cells > 0" and "pointer.ptr.free_cells == 0" will depend on updated pointer.  So whether the memory order should be memory_order_acquire while compare_exchange_weak return failure. 
That is
```c++
        while (pointer.ptr.free_cells > 0 &&
                !node_->pointer_.compare_exchange_weak(pointer,
                { { inc_pointer(pointer.ptr.write_p), pointer.ptr.free_cells - 1 } },
                std::memory_order_release,
                std::memory_order_acquire))
```

CC: @fujitatomoya This is my thought.

MiguelCompany, 2024-11-21:
Looking at the following example from the [std::memory_order](https://en.cppreference.com/w/cpp/atomic/memory_order) documentation makes me think that we only need the changes suggested by @fujitatomoya in https://github.com/ros2/rmw_fastrtps/issues/789#issuecomment-2489405226

```c++
#include <atomic>
#include <cassert>
#include <thread>
#include <vector>
 
std::vector<int> data;
std::atomic<int> flag = {0};
 
void thread_1()
{
    data.push_back(42);
    flag.store(1, std::memory_order_release);
}
 
void thread_2()
{
    int expected = 1;
    // memory_order_relaxed is okay because this is an RMW,
    // and RMWs (with any ordering) following a release form a release sequence
    while (!flag.compare_exchange_strong(expected, 2, std::memory_order_relaxed))
    {
        expected = 1;
    }
}
 
void thread_3()
{
    while (flag.load(std::memory_order_acquire) < 2)
        ;
    // if we read the value 2 from the atomic flag, we see 42 in the vector
    [assert](http://en.cppreference.com/w/cpp/error/assert)(data.at(0) == 42); // will never fire
}
 
int main()
{
    [std::thread](http://en.cppreference.com/w/cpp/thread/thread) a(thread_1);
    [std::thread](http://en.cppreference.com/w/cpp/thread/thread) b(thread_2);
    [std::thread](http://en.cppreference.com/w/cpp/thread/thread) c(thread_3);
    a.join(); b.join(); c.join();
}
```

Barry-Xu-2018, 2024-11-21:
````c++
        Cell* head()
        {
            auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);

            // If local read_pointer and write_pointer are equal => buffer is empty for this listener
            if (read_p_ == pointer.ptr.write_p )
            {
                return nullptr;
            }

            auto cell = &buffer_.cells_[get_pointer_value(read_p_)];

            return cell->ref_counter() != 0 ? cell : nullptr;
        }
````
If memory order is `memory_order_acquire`,  it can make sure reading operations (such as read_p_ == pointer.ptr.write_p) must be after `auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);`

if memory order is `memory_order_relaxed`, reading operations may be done before `auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);` 
But I think `read_p_ == pointer.ptr.write_p` have to be located after `auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);`  since pointer is a temporary variable.  But `auto cell = &buffer_.cells_[get_pointer_value(read_p_)];` may be done before `auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);`   . Even if executed in this order, the final output won't be affected.  

Of course, to ensure order, I agree with changing to use `memory_order_acquire`.

fujitatomoya, 2024-11-21:
@Barry-Xu-2018 

i may be mistaken on some points...

> If memory order is memory_order_acquire, it can make sure reading operations (such as read_p_ == pointer.ptr.write_p) must be after auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);

not really sure what this means here.
i think it just make sure all the store operation in other threads where calls `release` can be visible on this thread here.

> if memory order is memory_order_relaxed, reading operations may be done before auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);

this should not be happening, and compiler should not do that. if it does, it breaks the program sequence.

Barry-Xu-2018, 2024-11-22:
@fujitatomoya 

> not really sure what this means here.
> i think it just make sure all the store operation in other threads where calls release can be visible on this thread here.

I want to express that no reads or writes in the current thread can be reordered before this load.
Refer to the explanation of memory_order_acquire in https://en.cppreference.com/w/cpp/atomic/memory_order.

>> if memory order is memory_order_relaxed, reading operations may be done before auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);
>
> this should not be happening, and compiler should not do that. if it does, it breaks the program sequence.

`memory_order_relaxed`  only guarantees that the operation is atomic.  Due to compiler or execution optimization, it's possible that "auto cell = &buffer_.cells_[get_pointer_value(read_p_)];" might be moved ahead of "buffer_.node_->pointer_.load()".  
`memory_order_acquire` can avoid this optimization.

```c++
        Cell* head()
        {
            auto pointer = buffer_.node_->pointer_.load(std::memory_order_relaxed);

            // If local read_pointer and write_pointer are equal => buffer is empty for this listener
            if (read_p_ == pointer.ptr.write_p )
            {
                return nullptr;
            }

            auto cell = &buffer_.cells_[get_pointer_value(read_p_)];

            return cell->ref_counter() != 0 ? cell : nullptr;
        }
````

MiguelCompany, 2024-11-29:
@fujitatomoya Are you going to try the patch proposed in https://github.com/ros2/rmw_fastrtps/issues/789#issuecomment-2489405226?

If you see improvements with that, I would then think of a regression test, and prepare a PR for Fast DDS with that

fujitatomoya, 2024-12-02:
@MiguelCompany we are still working on how situation changes with the patch and UDP transport, let me have a couple of days. thanks for checking.

fujitatomoya, 2024-12-03:
@MiguelCompany 

- UDP Transport (w/o SHM transport)
  - the problem (tf dropping message) can be reproduced, that tells us this problem is not only with SHM transport.
  - when the problem starts happening [eprosima::fastrtps::rtps::StatefulReader::processGapMsg](https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/messages/MessageReceiver.cpp#L1230) consumes CPU time a lot compared to normal situation. This indicates that publisher skips or drops the certain messages on the HistoryCache, and prevent subscriber from request the retransmission for the efficiency.

- [patch](https://github.com/ros2/rmw_fastrtps/issues/789#issuecomment-2489405226) applied with SHM transport
  - can mitigate situation until the spike. w/o patch it was 10 hours, but with patch it can be more than 35 hours.
  - even before the spike, [Buffer Full situation](https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp#L228-L231) are appeared in order of 10~100 constantly. this is different from without patch condition.​
  - once it hits the spike, there are hundreds thousands [Buffer Full situation](https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp#L228-L231) are appeared... almost same without patch situation.

i believe that [Buffer Full situation](https://github.com/eProsima/Fast-DDS/blob/b1e4707ad3cfe4cad7e5100318402206e0bd5e78/src/cpp/rtps/transport/shared_mem/MultiProducerConsumerRingBuffer.hpp#L228-L231) should not be true in the 1st place...

fujitatomoya, 2024-12-03:
CC: @Barry-Xu-2018

MiguelCompany, 2024-12-04:
@fujitatomoya 

From your description on https://github.com/ros2/rmw_fastrtps/issues/789#issuecomment-2513579730, this seems like a slowly growing snowball issue.

It seems that from time to time, independently of the transport used, there are packet drops due to the transport buffers (UDP or SHM) being full. That would make the writer resend some data, making the snowball grow and leading to the buffers filling more quickly (since it now has to send both old and new data).

The presence of gap messages is evidence that the history of the writer became full, and some samples were removed from the history before they were acknowledged by the readers.

In this situation, increasing the buffers in UDP, or increasing both segment size and port capacity in SHM would improve the situation.

My suggestion would be to start with UDP only and increase the buffers till you make the situation stable. Then add SHM (with the patch), and use the same size for the segments as you used for the UDP buffers

fujitatomoya, 2024-12-04:
@MiguelCompany 

> It seems that from time to time, independently of the transport used, there are packet drops due to the transport buffers (UDP or SHM) being full. That would make the writer resend some data, making [...]

totally agree.

> My suggestion would be to start with UDP only and increase the buffers till you make the situation stable. Then add SHM (with the patch), and use the same size for the segments as you used for the UDP buffers

we kinda already tried with `port_queue_number`.
the bigger `port_queue_number` is, the longer time until we hit this snow ball issue. (e.g `128` 10 hours, `512` it takes 37 hours (default))
but eventually, we will hit this issue...

the original requirement is that robots at site should be 24/7 running without power-off.
i think that is common requirement if that is factory and logistics for robot automation... but with this issue, we cannot support that requirement.

but we have not tried with UDP send/receive `buffer size` and SHM transport `segment size` yet. we will give it a shot how that changes the situation.

by the way, there is a general question about [Data Sharing delivery](https://fast-dds.docs.eprosima.com/en/v3.0.0/fastdds/transport/datasharing.html) and [SHM transport](https://fast-dds.docs.eprosima.com/en/v3.0.0/fastdds/transport/shared_memory/shared_memory.html).

with SHM transport, descriptors are stored in the port with mutex, and that descriptor describes where to access the data where is shared memory address of the participants who send the data.
and this shared memory segment is allocated based on the configuration file, in default it says `512*1024` byte.
if the data sharing delivery is not enabled, it sounds reasonable that configuration of this segment size affects the `buffer full` situation.

what if the data sharing delivery is enabled? i mean in this case, DataWriter allocates the shared memory segment based on the QoS configuration history and depth as in HistoryCache, and that is coupled with DataReader.
so in data sharing delivery, this DataWriter's HistoryCache shared memory segment is used by SHM transport segment as well to achieve zero copy delivery?
that said, if data sharing delivery is enabled, we need to configure QoS for DataWriter, not SHM transport segment size?

i am trying to comprehend the memory management with data sharing delivery and SHM transport both enabled.

fujitatomoya, 2024-12-05:
> but we have not tried with UDP send/receive buffer size and SHM transport segment size yet. we will give it a shot how that changes the situation.

we have tried with SHM segment size, but it does not solve the root cause... it just extends the time window until the problem happens. (current segment size is 10MB, but we can still meet the problem.)

MiguelCompany, 2024-12-09:
@fujitatomoya One thing you could try is reducing the heartbeat period on the publishers. You can do that by XML, see [this section](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/datawriter.html#datawriter-configuration) of the documentation.

Reducing the `heartbeat_period` would make the reliability mechanism more responsive, thus (hopefully) reducing the possibility of old and new data to be sent at the same time. I would start with something like heartbeat_period = 0.7 * publication_period.

fujitatomoya, 2024-12-09:
@MiguelCompany 

could you share your thoughts on https://github.com/ros2/rmw_fastrtps/issues/789#issuecomment-2518064377

> by the way, there is a general question about [Data Sharing delivery](https://fast-dds.docs.eprosima.com/en/v3.0.0/fastdds/transport/datasharing.html) and [SHM transport](https://fast-dds.docs.eprosima.com/en/v3.0.0/fastdds/transport/shared_memory/shared_memory.html).

i am not sure if this is related, but would like to know how Fast-DDS deals with this configuration.

fujitatomoya, 2024-12-09:
@MiguelCompany thanks!

> Reducing the heartbeat_period would make the reliability mechanism more responsive, thus (hopefully) reducing the possibility of old and new data to be sent at the same time. I would start with something like heartbeat_period = 0.7 * publication_period.

will give it a shot.

MiguelCompany, 2025-01-03:
> will give it a shot.

@fujitatomoya Any updates here?

fujitatomoya, 2025-01-03:
@MiguelCompany no not yet, sorry.

teams in Tokyo will be back in next week, we will let you know once we got any updates!

fujitatomoya, 2025-01-13:
@MiguelCompany 

sorry for being late to get back to this, finally we have the evaluation result.

the outcome does not really change, `buffer full` on SHM transport problem still stays. (actually it became worse...)

> Reducing the heartbeat_period would make the reliability mechanism more responsive, thus (hopefully) reducing the possibility of old and new data to be sent at the same time. I would start with something like heartbeat_period = 0.7 * publication_period.

the following result with https://github.com/eProsima/Fast-DDS/pull/5540.

based on our publish rate, it set to `17.5 msec` (40Hz * 70%).
Changing heartbeat_period was not effective, `Buffer full` logs appeared many times right after launching.
besides that CPU consumption increases up to 90%, compared to before 40~50%.

my assumption was that we could expect the certain extra CPU consumption, since this configuration adds the heartbeat message more to ungap the condition between DataWriter and DataReader. and this could mitigate the `Buffer full` situation.

but it does not look like it works in that way at all, it actually increases CPU a lot and then that gets the situation much worse to introduce the `Buffer full` problem.

fujitatomoya, 2025-01-22:
@MiguelCompany any update on this?

besides above result, we also checked the application callback interval.
the application callback should be called every 0.025 seconds (40Hz), but when the problem happens it can be more than 2 seconds to be called for the next callback.

fujitatomoya, 2025-02-13:
@MiguelCompany we are trying to set up the reproducible example now, as we talked before.

fujitatomoya, 2025-03-07:
@MiguelCompany we can  reproduce this issue with the following patches applied.

originally happens in humble, but today can be observed with today's source build https://github.com/ros2/ros2/commit/fa63fcfb056136c3950939ced225313c95198a3d

1. apply patches https://github.com/ros2/examples and https://github.com/eProsima/Fast-DDS/tree/3.1.x
2. rebuild the workspace
3. `ros2 launch examples_rclcpp_minimal_subscriber test.launch.xml`
4. the following `Buffer full` messages are on the console.

```bash
...<snip>
[subscriber_lambda-43] subscriber_lambda, Buffer full. Fri Mar  7 11:49:44 2025
[subscriber_lambda-43] subscriber_lambda, Buffer full. Fri Mar  7 11:49:44 2025
[subscriber_lambda-43] subscriber_lambda, Buffer full. Fri Mar  7 11:49:44 2025
[publisher_lambda-12] publisher_lambda, Buffer full. Fri Mar  7 11:49:44 2025
[publisher_lambda-12] publisher_lambda, Buffer full. Fri Mar  7 11:49:44 2025
[publisher_lambda-12] publisher_lambda, Buffer full. Fri Mar  7 11:49:44 2025
...<snip>
```

40Hz on `topic` with `std_msgs::msg::String` type. (Non-PoD, data sharing cannot be enabled.)
30 publishers x 30 subscriptions: can reproduce the many `Buffer full` messages on the publishers, sometimes subscribers too during data communication.
with my local environment 25 x 25 case: it does not reproduce this `Buffer full` messages.

<details><summary>examples patch</summary>

apply this patch to https://github.com/ros2/examples

```examples.patch
iff --git a/rclcpp/topics/minimal_publisher/lambda.cpp b/rclcpp/topics/minimal_publisher/lambda.cpp
index 324eb96..b84215e 100644
--- a/rclcpp/topics/minimal_publisher/lambda.cpp
+++ b/rclcpp/topics/minimal_publisher/lambda.cpp
@@ -36,10 +36,10 @@ public:
       [this]() -> void {
         auto message = std_msgs::msg::String();
         message.data = "Hello, world! " + std::to_string(this->count_++);
-        RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", message.data.c_str());
+        //RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", message.data.c_str());
         this->publisher_->publish(message);
       };
-    timer_ = this->create_wall_timer(500ms, timer_callback);
+    timer_ = this->create_wall_timer(25ms, timer_callback);
   }

 private:
diff --git a/rclcpp/topics/minimal_subscriber/CMakeLists.txt b/rclcpp/topics/minimal_subscriber/CMakeLists.txt
index 489fd5b..fd7ae98 100644
--- a/rclcpp/topics/minimal_subscriber/CMakeLists.txt
+++ b/rclcpp/topics/minimal_subscriber/CMakeLists.txt
@@ -55,6 +55,11 @@ rclcpp_components_register_node(wait_set_subscriber_library
     PLUGIN "TimeTriggeredWaitSetSubscriber"
     EXECUTABLE time_triggered_wait_set_subscriber)

+install(DIRECTORY
+  launch
+  DESTINATION share/${PROJECT_NAME}
+)
+
 install(TARGETS
     wait_set_subscriber_library
     ARCHIVE DESTINATION lib
diff --git a/rclcpp/topics/minimal_subscriber/lambda.cpp b/rclcpp/topics/minimal_subscriber/lambda.cpp
index 16b9083..4c35a01 100644
--- a/rclcpp/topics/minimal_subscriber/lambda.cpp
+++ b/rclcpp/topics/minimal_subscriber/lambda.cpp
@@ -12,6 +12,8 @@
 // See the License for the specific language governing permissions and
 // limitations under the License.

[... 103 lines cut ...]
 #include <cstdlib>

+#include <iostream>
+#include <cstdlib>
+#include <chrono>
+#include <ctime>
+#include <sys/time.h>
+extern char *__progname;
+
 namespace eprosima {
 namespace fastdds {
 namespace rtps {
@@ -227,6 +234,9 @@ public:

         if (pointer.ptr.free_cells == 0)
         {
+            auto now = std::chrono::system_clock::now();
+            std::time_t time =std::chrono::system_clock::to_time_t(now);
+            std::cerr << std::string(__progname) << ", "  << "Buffer full. " << std::ctime(&time);
             throw std::runtime_error("Buffer full");
         }

```

</details>

fujitatomoya, 2025-04-21:
@MiguelCompany do you have any thoughts or anywhere we could look into with reproducible example?

fujitatomoya, 2025-05-13:
@MiguelCompany what do you think about this? possible undetected bug?

according to our analysis,

- https://github.com/eProsima/Fast-DDS/blob/e7510c91c42489b264201e156ff8ccf77d1e4117/src/cpp/rtps/transport/shared_mem/SharedMemChannelResource.hpp#L184-L201: s the time duration to pop the message is almost always same even with this buffer full condition, but frequency to call on `pop` is decreasing gradually.
- on the flip side, we can observe the elapsed time for `MessageReceiver`, https://github.com/eProsima/Fast-DDS/blob/e7510c91c42489b264201e156ff8ccf77d1e4117/src/cpp/rtps/network/ReceiverResource.cpp#L135 takes more time gradually. (this can be the reason for the frequency decreasing.)
- it turned out that `proc_Submsg_Gap` elapsed time increases gradually, and eventually this is stopped to be called after 40 hours run test. (we believe that something left behind in the buffer cannot be taken out in particular situation, and that leads to this racy condition. eventually buffer will be fill with garbage, that stops calling `proc_Submsg_Gap`.

why `proc_Submsg_Gap` elapsed time increases gradually? `StatefulReader::processGapMsg` takes more time to process, our assumption is that it takes more time to check the sequence numbers in the gapList, and filter them out from the gapList since those are already irrelevant. in other words, gapList `finalAN` is getting bigger and that makes the loop much many times in this function.
And we have confirmed that `finalAN` increases gradually more than 20,000 as system runs. https://github.com/eProsima/Fast-DDS/blob/e7510c91c42489b264201e156ff8ccf77d1e4117/src/cpp/rtps/reader/StatefulReader.cpp#L936 (this proves our assumption is correct above.)

then the next question is why `StatefulWriter` sends these huge gapList to `StatefulReader` anyway?

we also confirmed that in `StatefulWriter::add_gaps_for_holes_in_history`, `firstSeq` is always 1.
that means, `(lastSeq - HistorySize)` (lastSeq becomes huge as data communication keeps running) is the gapList to be sent to the concerned `DataReader`.
https://github.com/eProsima/Fast-DDS/blob/e7510c91c42489b264201e156ff8ccf77d1e4117/src/cpp/rtps/writer/StatefulWriter.cpp#L2249

this is really inconsistent condition, since all readers have already acked the sequence number 1, but it accidentally stays in the Writer History forever. (this is the reason that GAP messages are generating a lot.)
we think that `StatefulWriter::check_acked_status` misses the racy condition that there are ack required reader and not required reader exists? (our reproducible test program and our application, many reader joins and leaves during constantly.)
so far, the following patch fixes the problem with our long run test in the simulation environment.

```diff
diff --git a/src/ros2/fastrtps/src/cpp/rtps/writer/StatefulWriter.cpp b/src/ros2/fastrtps/src/cpp/rtps/writer/StatefulWriter.cpp
index 78459ef3c5d..9c18e75e7a9 100644
--- a/src/ros2/fastrtps/src/cpp/rtps/writer/StatefulWriter.cpp
+++ b/src/ros2/fastrtps/src/cpp/rtps/writer/StatefulWriter.cpp
@@ -1424,6 +1424,14 @@ void StatefulWriter::check_acked_status()
             }
             );
 
+    // In situations where there are recently matched readers to which the writer has
+    // not yet sent changes, both min_low_mark = zero and all_acked == true will be
+    // met. In this scenario, onWriterChangeReceivedByAll() will be skipped for the
+    // acked changes, causing the changes to remain in history indefinitely. To
+    // prevent this, adjust min_low_mark to handle the acked changes.
+    if (all_acked)
+        min_low_mark = mp_history->next_sequence_number() - 1;
+
     bool something_changed = all_acked;
     SequenceNumber_t min_seq = get_seq_num_min();
     if (min_seq != SequenceNumber_t::unknown())
```

if we are correct, this is RTPS issue. so that makes sense we can reproduce this issue with UDP (not SHM transport) as well.

fujitatomoya, 2025-05-15:
@MiguelCompany this problem has been confirmed with rolling, so i do not think https://github.com/eProsima/Fast-DDS/pull/5606 can fix our problem.

besides that, https://github.com/eProsima/Fast-DDS/pull/5606 only avoids failing `try_remove_change` when the history cache is empty. i think that is the different situation described above, the problem clips the `min_low_mark` with zero when the DataReaders are joining at the same time. i think if it is determined as `all acked`, we can safely set the `min_low_mark` into current sequence number to remove old data because those are acknowledged already?

MiguelCompany, 2025-05-16:
> `StatefulReader::processGapMsg` takes more time to process, our assumption is that it takes more time to check the sequence numbers in the gapList, and filter them out from the gapList since those [...]

@fujitatomoya 
Not sure about the writer side (need some time to analyse), but the reader side can be optimized by making the loop start in `max(gapStart, pWP->available_changes_max())`

fujitatomoya, 2025-05-16:
@MiguelCompany just in case we tried https://github.com/eProsima/Fast-DDS/pull/5606 with humble, and the problem can be still observed.

we might be missing something, but our analysis tells me the root cause of this is that change in the history cache of writer side will never be removed, and that generates a lot of GAP messages as time goes by, eventually buffer full. (we checked this condition with logging data, and that is what happens as far as we see.) and this happens with UDP too.
this problem requires long run for days, depending on number of DataReader and publication frequency. but once this happens, there is no way around unless reboot the DDS network. i think this is kind of critical issue for stability and reliability... anyway, if you need any other information, or find out the insight, please let me know. we are doing our test with suggested patch, so far the problem has never been reproduced.

> Not sure about the writer side (need some time to analyse), but the reader side can be optimized by making the loop start in max(gapStart, pWP->available_changes_max())

this makes sense to bounce up the starting number to actual available change max instead of gapStart, right? this actually can mitigate the processing time on reader side when certain amount of GAP messages sent by writer. but i do not think this can fix our problem, because the root cause of the writer side.

fujitatomoya, 2025-05-21:
@MiguelCompany 

the application uses geometry2, related QoS configuration is https://github.com/ros2/geometry2/blob/humble/tf2_ros/include/tf2_ros/qos.hpp.
i am double-checking this but i believe they use default setting, that means KEEP_LAST (rclcpp::QoS default) and Depth with 100.

our assumption here is that, Writer's History does not get full with capacity 100 entries, but old samples just stay in the History cache, that leads to this GAP message explosion situation.

fujitatomoya, 2025-05-28:
I say it is `DynamicBroadcasterQoS`, but just in case let me check with team members.

fujitatomoya, 2025-06-09:
@MiguelCompany i just go ahead to create the PR on https://github.com/eProsima/Fast-DDS/pull/5859