ros2/rmw_fastrtps #793: Cyclone DDS and Fast RTPS exhibiting vastly different CPU usage when running same ROS2 application on Jetson Xavier AGX SoC

nat-tsang, 2024-12-15:
**Required Info:**
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - From source
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rclcpp

#### Expected behavior
In comparing our processes running on rmw_fastrtps_cpp and rmw_cyclonedds_cpp, the processes running with fastrtps cause all 8 of our CPU threads on the Jetson Xavier AGX to run at maximum usage 95-100% which is causing there to be no topics listed, and preventing the processes from publishing/subscribing. With cyclone_dds, the exact same process can run as expected, with all of the threads of the CPU running at around 15-20% usage. We have tested running the processes with both middlewares for up to 48 hours, and after 48 hours the cyclonedds process can still run normally whereas the fastrtps process will reach <100% CPU usage in the hour. We are wondering if this has been seen before, and what might be the root cause of the differences in operation of fastrtps and cyclonedds.

nat-tsang, 2024-12-16:
Hi! We are using ROS2 Foxy which comes with Fast RTPS 2.1.0. We were using the default version, so I will try disabling Shared Memory to see if the issue persists

cferreiragonz, 2024-12-17:
Hi @nat-tsang, please note that ROS 2 Foxy has reached its End of Life (EOL) more than a year ago, and the same applies to Fast RTPS 2.1. If disabling Shared Memory does not resolve the issue, could you consider upgrading to ROS 2 Humble, which uses Fast RTPS 2.6? Testing with these newer versions may help solve the issue if it persists.

nat-tsang, 2024-12-17:
Hi, I have found that disabling Shared Memory has been effective so far in preventing the CPU from throttling as the application has been running for the last 48 hours without throttling the CPU. Why does disabling Shared Memory produce better CPU efficiency? What exactly is happening when Shared Memory is disabled?

cferreiragonz, 2024-12-18:
[Shared Memory](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/shared_memory/shared_memory.html#shared-memory-transport) (SHM) is one of the default transports used by Fast DDS, alongside UDP. It is automatically employed whenever two entities are on the same host, as it typically provides better performance for data transmission in terms of memory usage and latency. However, the architecture of SHM differs from that of UDP, and this difference can impact CPU thread behavior. For example, using SHM generally results in fewer blocking calls on the receiver side, which can affect CPU usage. The extent of this impact varies depending on factors such as the kernel, operating system, and hardware configuration.

By disabling SHM, you rely exclusively on UDP for data transmission. While this may be less resource-intensive in some scenarios, it can result in slightly slower throughput performance. The best option depends on your specific use case and setup requirements.

MiguelCompany, 2025-01-03:
Closing since all the necessary information has been provided

closed 2025-01-03 by MiguelCompany