ros2/rmw_fastrtps #796: ros2 node list issue on foxy

fire9953, 2025-01-15:
## 

**Required Info:**

- Operating System:
  - Ubuntu 20.04，ros2-foxy
- DDS implementation:
  - ros-foxy-rmw-fastrtps-cpp-dbgsym/focal 1.3.2-1focal.20230527.064419 arm64
ros-foxy-rmw-fastrtps-cpp/focal,now 1.3.2-1focal.20230527.064419 arm64 [installed,automatic]
ros-foxy-rmw-fastrtps-dynamic-cpp-dbgsym/focal 1.3.2-1focal.20230527.064406 arm64
ros-foxy-rmw-fastrtps-dynamic-cpp/focal 1.3.2-1focal.20230527.064406 arm64
ros-foxy-rmw-fastrtps-shared-cpp-dbgsym/focal 1.3.2-1focal.20230527.052644 arm64
ros-foxy-rmw-fastrtps-shared-cpp/focal,now 1.3.2-1focal.20230527.052644 arm64 [installed,automatic]

#### Steps to reproduce issue
I‘m working on ros2-foxy , i often find that when using "ros2 node list", I can't see the nodes, or can only see a part of them. I've checked some posts here and found that upgrading FastDDS might solve this problem?  

How should I upgrade FastDDS? I'm currently working on Foxy. Which version should I upgrade to?  How to upgrade it?
I'm really confused right now and need some help.

fujitatomoya, 2025-01-15:
@fire9953 thanks for creating issue.

Foxy is already End of Life, can you upgrade either humble or jazzy to see if you can still meet this issue?

fire9953, 2025-01-16:
@fujitatomoya thanks，but i'm working on jetpack 5.1.x now，it is based on ubuntu 20.04, it's complex to upgrade foxy to humble. I read this post, which mentioned "fixed by the eProsima/Fast-DDS#3753", so I wonder if I can upgrade my current DDS version. BUT Which version of DDS can be used in the Foxy version? I'm a bit confused. Please give me some advice. Thank you.

fujitatomoya, 2025-01-16:
> BUT Which version of DDS can be used in the Foxy version?

for foxy, Fast-DDS version was `2.1.x`.

https://github.com/ros2/ros2/blob/3bc9b4cd9a6661f09ae5f271d4f5f4a3b11cccd9/ros2.repos#L38-L41

and https://github.com/eProsima/Fast-DDS/pull/3753 is not backported to `2.1.x`. (i am not even sure if the patch can be applied.)

fujitatomoya, 2025-01-16:
this is likely for the users especially for BSP. i would try container instead of running the application on the host system.

if you do not have any other questions, can you close this issue?

MiguelCompany, 2025-01-16:
@fire9953 I would try building the following repos from source:

```
repositories:
  eProsima/Fast-CDR:
    type: git
    url: https://github.com/eProsima/Fast-CDR.git
    version: v1.0.24
  eProsima/Fast-DDS:
    type: git
    url: https://github.com/eProsima/Fast-DDS.git
    version: v2.6.9
  ros2/rmw_fastrtps:
    type: git
    url: https://github.com/ros2/rmw_fastrtps.git
    version: 1.3.2
```

fire9953, 2025-01-17:
@MiguelCompany  Thanks very much , I got some hints from this link https://github.com/ros2/ros2cli/issues/582 . 
I want to confirm with you , Can upgrading DDS solve the problem of node loss?

clalancette, 2025-01-30:
While building a newer version of Fast-DDS may help here, it is probably easiest to upgrade all of ROS 2 rather than cherry-picking some repositories.  At the very least, we test an entire distribution together.

Towards that end, I may suggest that you build Humble from source on Ubuntu 20.04.  Instructions on how to do so are at https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

Since this question is for an End-of-Life distribution, I'm going to go ahead and close this out.

closed 2025-01-30 by clalancette