ros2/rmw_fastrtps #798: Specifying interfaceWhitelist does not work

rty813, 2025-01-20:
I have two hosts, each with two interconnected network interfaces: eth0 and edge0. I want them to communicate via the eth0 interface rather than the edge0. However, after specifying the `interfaceWhiteList`, traffic still flows through both network interfaces simultaneously.

--- 

NICs information:
||Host1|Host2|
|:---:|:---:|:---:|
|eth0|192.168.3.11/24|192.168.3.112/24|
|edge0|10.1.233.165/16|10.1.123.79/16|

---

fastrtps xml:
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>CustomUDPTransport</transport_id>
            <type>UDPv4</type>
            <interfaceWhiteList>
                <address>192.168.3.11</address>
                <address>127.0.0.1</address>
            </interfaceWhiteList>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="CustomUDPTransportParticipant">
        <rtps>
            <useBuiltinTransports>false</useBuiltinTransports>
            <userTransports>
                <transport_id>CustomUDPTransport</transport_id>
            </userTransports>
        </rtps>
    </participant>
</profiles>
```

---

`tcpdump -i edge0 udp`:
```
......
15:28:26.397300 IP 10.1.233.165.46534 > 10.1.123.79.7410: UDP, length 68
15:28:26.400399 IP 10.1.233.165.46534 > 10.1.123.79.7410: UDP, length 68
15:28:26.403855 IP 10.1.233.165.46534 > 10.1.123.79.7410: UDP, length 68
15:28:26.408088 IP 10.1.233.165.46534 > 10.1.123.79.7410: UDP, bad length 20652 > 1256
15:28:26.411637 IP 10.1.233.165 > 10.1.123.79: udp
15:28:36.577914 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:36.778463 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:36.979111 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:37.178606 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:37.379147 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:37.579511 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:37.780080 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:37.980124 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:38.022211 IP 10.1.233.165.46534 > 10.1.123.79.7410: UDP, length 392
15:28:38.028020 IP 10.1.233.165.34813 > 239.255.0.1.7400: UDP, length 392
15:28:38.180172 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
15:28:38.380327 IP 10.1.233.165.46534 > 10.1.123.79.7411: UDP, length 360
......
```

`tcpdump -i eth0 udp`:
```
......
15:32:59.227396 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
15:32:59.427499 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
15:32:59.627599 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
15:32:59.828124 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
15:33:00.028431 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
15:33:00.228654 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
15:33:00.429054 IP 192.168.3.11.46534 > 192.168.3.112.7411: UDP, length 360
......
```

MiguelCompany, 2025-01-20:
@rty813 In order for the participant configuration to be taken into account, the profile should have `is_default_profile="true"` attribute.

```
<participant profile_name="CustomUDPTransportParticipant" is_default_profile="true">
```

rty813, 2025-01-21:
@MiguelCompany Adding `is_default_profile` did indeed fix the issue. However, I'm encountering a new problem now. If I specify `initialPeerList`, the communication between both sides fails, but if I remove it, the communication works fine. What could be the reason for this?

```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <participant profile_name="CustomUDPTransportParticipant" is_default_profile="true">
        <rtps>
            <builtin>
                <metatrafficUnicastLocatorList>
                    <locator/>
                </metatrafficUnicastLocatorList>
                <initialPeersList>
                    <locator>
                        <udpv4>
                            <address>192.168.3.11</address>
                            <address>192.168.3.112</address>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
        </rtps>
    </participant>
</profiles>
```

rty813, 2025-01-21:
There's another issue. When both UDP and SHM are enabled on the same host, how does Fast RTPS choose between them? I see that both are transmitting data.

rty813, 2025-01-21:
> [@rty813](https://github.com/rty813) In order for the participant configuration to be taken into account, the profile should have `is_default_profile="true"` attribute.
> 
> ```
> <participant profile_name="CustomUDPTransportParticipant" is_default_profile="true">
> ```

I tested it, and now the XML configuration seems to be effective, but there is still some traffic going through the `edge0` network interface. If I keep only 127.0.0.1 in the whitelist, I can ensure that no messages are sent through `edge0`. However, if I add the IP of eth0, the messages are forwarded through both eth0 and edge0 simultaneously.

some traffic log:
```
15:50:28.258271 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 252
15:50:28.262546 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 68
15:50:28.265470 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 68
15:50:28.268201 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 68
15:50:28.271507 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 460
15:50:28.275220 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 460
15:50:28.278129 IP 192.168.3.112.48183 > 10.1.233.164.7410: UDP, length 432
```

MiguelCompany, 2025-01-21:
If you want to use the same XML in both hosts, put all the addresses you want to whitelist, i.e.:

```
            <interfaceWhiteList>
                <address>192.168.3.11</address>
                <address>192.168.3.112</address>
                <address>127.0.0.1</address>
            </interfaceWhiteList>
```

MiguelCompany, 2025-01-21:
In ROS 2 Jazzy, you can also whitelist interface names:

```
            <interfaceWhiteList>
                <interface>eth0</interface>
                <interface>lo</interface>
            </interfaceWhiteList>
```

rty813, 2025-01-21:
> If you want to use the same XML in both hosts, put all the addresses you want to whitelist, i.e.:
>
> ```
>             <interfaceWhiteList> [...]

I didn't use the same XML for both hosts, although I did try adding both IPs, but there is still traffic being transmitted through the edge0 network card.

TZECHIN6, 2025-05-24:
> [@rty813](https://github.com/rty813) In order for the participant configuration to be taken into account, the profile should have `is_default_profile="true"` attribute.
> 
> ```
> <participant profile_name="CustomUDPTransportParticipant" is_default_profile="true">
> ```

Hi, I am also working on specifc the DDS to use the ethernet interface only. The reason is that my ubuntu 22 system have a wifi chip MT7929 which is currently not being supported by the kernel. I would like to ensure all the ROS2 communication is going via the ethernet not the WiFi interface. From the document https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/interfaces.html#ifaces-config, there is no line `is_default_profile="true"`. Should we still put this in the line?

cferreiragonz, 2025-05-26:
> Should we still put this in the line?

Yes. You should use the `is_default_profile="true"` to configure whitelist in ROS 2. 

It is mentioned in Fast DDS documentation specific for ROS 2: https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html#creating-ros-contexts-and-nodes

rty813, 2025-07-01:
The previous test result was wrong. After retesting, it works fine. Close the issue