ros2/rmw_fastrtps #800: Support: Static EDP with one publisher and two subscribers DDS and ROS2

Syamkamalnadhc, 2025-01-22:
I referred the example code in FastDDS StaticHelloWorld example DDS application for DDS and the below link for ROS2 application

Link: https://github.com/ros2/rmw_fastrtps/issues/617

I am using the same setup mentioned in the link (Ros2 humble and ubuntu 22.04), I tried with UDP only case and it seems working for single publisher and subscriber configuration.

Now I am facing issue when configuring single publisher and two subscribers with staticHelloWorld example DDS application

The HelloworldPublisher xml used, and loaded in two subscribers

[image]

The HelloworlSubscriber xml  used and loaded in the publisher( expected to publish to both subscribers)

[image]

Here two readers are configured and don't know this is the right way...

Questions
1. How to configure for multiple subscriber combination
2. Do we have to configure other reader parameters (Guid  ) also for distinguishing readers

clalancette, 2025-01-22:
Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2025-01-22 by clalancette

Syamkamalnadhc, 2025-01-27:
Hi, I did not get any response yet, I have created a query there, please check

https://robotics.stackexchange.com/questions/114382/how-to-configure-static-edp-with-one-publisher-and-two-subscribers-dds-and-ros2