ros2/rmw_fastrtps #804: Question: Default is Synchronous Publishing, Right?

SteveMacenski, 2025-01-30:
Hi,

I'm putting together some educational materials on DDS and tuning and noticed some discrepencies in documentation.

https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html#changing-publication-mode in the table states that the default is `ASYNCHRONOUS_PUBLISH_MODE` in ROS 2, yet the [readme here](https://github.com/ros2/rmw_fastrtps?tab=readme-ov-file#change-publication-mode) says the opposite. 

I wanted to chime in to note this so that it could be resolved for the correct default by the maintainers / eProsima docs :-)

MiguelCompany, 2025-01-31:
@SteveMacenski The default is SYNCHRONOUS. We just merged https://github.com/eProsima/Fast-DDS-docs/pull/1025 fixing the Fast DDS documentation

fujitatomoya, 2025-01-31:
@MiguelCompany thanks! @SteveMacenski i will go ahead to close this one, if anything missing, feel free to reopen.

closed 2025-01-31 by fujitatomoya