ros2/rmw_fastrtps #807: How do I communicate FastDDS with ROS2's service

p-zzg, 2025-02-25:
How do I communicate FastDDS with ROS2's service
The topic can be prefixed with rt/ in the field on the name.
The service request is prefixed with rq/, and the reply is prefixed with rr/
You can send a message to the service, but you can't receive a reply

p-zzg, 2025-02-25:
Suffixes for requests and responses have also been added

fujitatomoya, 2025-02-25:
please open questions like this on [https://robotics.stackexchange.com/questions/tagged/ros](https://urldefense.com/v3/__https:/robotics.stackexchange.com/questions/tagged/ros__;!!O7_YSHcmd9jp3hj_4dEAcyQ!30VKjQJ3dEPBafWkOCyUZ2Aca-BlGQKmRnkhmldWkWOzl2yWsKh6xgy_3J0EKogYDFCk6iVW2bz87HKbgKHJhx1snGc$), which is our central Question and Answer site. You'll get a better answer there, and it will be searchable for the future.

closed 2025-02-25 by fujitatomoya