ros2/rmw_fastrtps #810: borrow_loaned_message() returns RETCODE_OUT_OF_RESOURCES when using Zero-copy

djusten, 2025-03-05:
I'm configuring my system to use Zero-copy and shared memory, also it publishes a custom msg that has old plain type, that means all fields with fixed size.
My test consist of one publisher and four subscribers. The fourth subscriber I keep stopping it, that means shutdown() is called and after few seconds it starts again. After few times, the publisher started to return a error when calling borrow_loaned_message(). One observation I had is the publisher "Crashes" when the subscriber is starting and not when it stopped.

Could be the subscriber somehow isn't releasing some loan? What could be causing the borrow_loaned_message() running out of resource if the callback is doing nothing, neither doing any copy or holding the data.

Running ROS2 humble.

**Publisher**
```
rclcpp::init(argc, argv);
rclcpp::executors::MultiThreadedExecutor exec{};
rclcpp::NodeOptions options = rclcpp::NodeOptions{};
[...]
const auto qos = rclcpp::QoS(10).history(rclcpp::HistoryPolicy::KeepLast).keep_last(10).reliable();
[...]
try {
  auto loanedMsg = publisher_->borrow_loaned_message();
  publisher_->publish(std::move(loanedMsg));
}
catch(const std::exception &e)
{
            std::cerr << "---------->" << e.what() << '\n';

}
[...]
exec.spin();
rclcpp::shutdown();
```

**Subscriber**
```
[...]
auto qosSub = rclcpp::QoS(10).history(rclcpp::HistoryPolicy::KeepLast).keep_last(10).reliable();
[...]
rclcpp::init(argc, argv);
rclcpp::executors::MultiThreadedExecutor exec{};
rclcpp::NodeOptions options = rclcpp::NodeOptions{};
[...]
void cb(const typename ros2_interface::msg::Image8m::UniquePtr msg)
{
//Doing nothing here;
}
[...]
exec.spin();
rclcpp::shutdown();
```

The environment variables are:
```
FASTRTPS_DEFAULT_PROFILES_FILE=/etc/ros_shm_profile.xml
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
RMW_FASTRTPS_USE_QOS_FROM_XML=1
ROS_DISABLE_LOANED_MESSAGES=0
ROS_LOCALHOST_ONLY=1
```

XML profile:
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <transport_descriptors>
        <!-- Create a descriptor for the new transport -->
        <transport_descriptor>
            <transport_id>shm_transport_only</transport_id>
            <type>SHM</type>
            <segment_size>6912512</segment_size>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="SHMParticipant" is_default_profile="true">
        <rtps>
            <!-- Link the Transport Layer to the Participant -->
            <userTransports>
                <transport_id>shm_transport_only</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
</profiles>
```

Actually `free_payloads_` is getting empty, but why if the subscriber is doing nothing with the data and why it isn't being released?

fujitatomoya, 2025-03-05:
IMO,

1st of all, you are not using [Data Sharing delivery](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/datasharing.html) nor [ROS 2 LoanedMessage](https://docs.ros.org/en/rolling/How-To-Guides/Configure-ZeroCopy-loaned-messages.html).
for humble, we need to explicitly enable data sharing to enable `LoandedMessage`, see https://github.com/ros2/rmw_fastrtps?tab=readme-ov-file#enable-zero-copy-data-sharing.

> borrow_loaned_message() returns RETCODE_OUT_OF_RESOURCES when using Zero-copy

i do not understand what this means.
`borrow_loaned_message()` returns `rclcpp::LoanedMessage<ROSMessageType, AllocatorT>`, what method actually returns `RETCODE_OUT_OF_RESOURCES`?

this problem seems to be related to [Shared Memory Transport](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/shared_memory/shared_memory.html), can you provide self-contained/complete example to reproduce the issue?

CC: @MiguelCompany

djusten, 2025-03-06:
Hi @fujitatomoya, thank you for your very quick reply.
I can't agree with what you mentioned above
> 1st of all, you are not using [Data Sharing delivery](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/datasharing.html) nor [ROS 2 LoanedMessage](https://docs.ros.org/en/rolling/How-To-Guides/Configure-ZeroCopy-loaned-messages.html).

As I mentioned above, my data type is [Plain Old Data](https://en.wikipedia.org/wiki/Passive_data_structure), the variables required `FASTRTPS_DEFAULT_PROFILES_FILE`, `RMW_FASTRTPS_USE_QOS_FROM_XML=1` and `ROS_DISABLE_LOANED_MESSAGES=0` are set as in the documentation. When using my main application (not this test app), I see the CPU load dropping down drastically when setting those variables, that means, it proofs the usage of zero-copy.
In case Zero-copy can't be used e.g. when NOT using Plain Old Data (e.g. sensor_msgs::msg::Image), the publisher generate a warning `[INFO] [1741251000.580629312] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.`, this is **NOT** happening, that means zero-copy is being used in my test application.
 Also using the following XML profile, I see the same behavior:
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <!-- Block gotten from https://fast-dds.docs.eprosima.com/en/v2.6.7/fastdds/transport/shared_memory/shared_memory.html#transport-sharedmemory-example -->
    <transport_descriptors>
        <!-- Create a descriptor for the new transport -->
        <transport_descriptor>
            <transport_id>shm_transport_only</transport_id>
            <type>SHM</type>
            <segment_size>6912512</segment_size>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="SHMParticipant">
        <rtps>
            <!-- Link the Transport Layer to the Participant -->
            <userTransports>
                <transport_id>shm_transport_only</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>

    <!-- Block gotten from https://github.com/ros2/rmw_fastrtps/tree/humble?tab=readme-ov-file#enable-zero-copy-data-sharing -->
    <!-- Default publisher profile -->
    <publisher profile_name="default publisher profile" is_default_profile="true">
        <qos>
            <data_sharing>
                <kind>AUTOMATIC</kind>
            </data_sharing>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </publisher>

    <!-- Default subscriber profile -->
    <subscriber profile_name="default subscriber profile" is_default_profile="true">
        <qos>
            <data_sharing>
                <kind>AUTOMATIC</kind>
            </data_sharing>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </subscriber>
</profiles>
```

> i do not understand what this means.
> borrow_loaned_message() returns rclcpp::LoanedMessage<ROSMessageType, AllocatorT>, what method actually returns RETCODE_OUT_OF_RESOURCES?

In case `borrow_loaned_message();` is inside a try{} catch (const std::exception &e), and borrow_loaned_message() fails it will throw `---------->error not set`, that doesn't help much, right? I went a bit forward and could see loan_sample() is returning OUT_OF_RESOURCE here https://github.com/eProsima/Fast-DDS/blob/2.6.7/src/cpp/fastdds/publisher/DataWriterImpl.cpp#L440.

My test consist in running 1 publisher and 4 subscriber. The forth subscriber I keep stopping (with CTRL+C) and running it again. After few times, the publisher stops sending data due no resource for loan.
Publisher
```
#include <chrono>
#include <functional>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "ros2_interface/msg/image8m.hpp"
#include "ros2_interface/msg/string.hpp"
#include <sensor_msgs/msg/image.hpp>
[... 118 lines cut ...]
        std::cout << "./main node_name topic" << std::endl;
        return -1;
    }
    rclcpp::init(argc, argv);

    rclcpp::executors::MultiThreadedExecutor exec{};
    rclcpp::NodeOptions options = rclcpp::NodeOptions{};

    // auto imageSubscriber = std::make_shared<Test>(options, argv[1]);
    auto imageSubscriber = std::make_shared<ImageSubscriber>(options, argv[1], argv[2]);
    exec.add_node(imageSubscriber);

    exec.spin();

    rclcpp::shutdown();

    std::cerr << ">>>>>>>>>>>>>>>>>>>>>>>>> shutdown success" << std::endl;
    imageSubscriber = nullptr;

    return 0;
}
```
Hope it could clarify a bit more.
Bests,
Diogo

djusten, 2025-03-21:
Hi @fujitatomoya, were you able to reproduce the issue? Is there anything else I can help on setting up the environment to reproduce it? Thanks.

fujitatomoya, 2025-03-23:
@djusten i could not allocate time, i will try in this week.

djusten, 2025-05-22:
Hi @fujitatomoya , we could workaround the issue in the past but now it is popping up with another use case. Basically if new subscribers are create with a certain frequency, the issue with borrow message in the publisher happens.
Did you have a chance to take a look on this? By any chance, do you have any new information to share?
Thank you again.
Bests, Diogo

djusten, 2025-10-28:
Hi @fujitatomoya, I would like to share the exact steps I'm following to simulate the issue.
Source code:  [pub_sub_examples.gz](https://github.com/user-attachments/files/23186092/pub_sub_examples.gz)

```
#terminal_1
tar -xf pub_sub_examples.tar
cd pub_sub_examples/
source ./env.sh
colcon build --build-base build_x86 --install-base install_x86 --parallel-workers 1 --packages-select sysmon
./install_x86/sysmon/lib/sysmon/publisher pub topic

#terminal_2
./install_x86/sysmon/lib/sysmon/subscriber sub1 topic

#terminal_3
./install_x86/sysmon/lib/sysmon/subscriber sub2 topic

#terminal_4
./install_x86/sysmon/lib/sysmon/subscriber sub3 topic

#terminal_5
./run.sh sub4
```

After few minutes (not many), you can see terminal_1 stops with error. Similar to:
```
publishing 10387
publishing 10388
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
---------->error not set
----------> EXIT
```

Any new insight is very welcome. Thank you again.
Bests, Diogo

SYIRPi, 2026-11-04:
Hi @djusten and @fujitatomoya,

I reproduced the loan failure on Ubuntu 22.04 / ROS 2 Humble with Fast DDS 2.6.12 and rmw_fastrtps_cpp 6.2.10. GDB observations suggest a missed history-cleanup notification in `StatefulWriter::check_acked_status()`.

My reduced test uses `std_msgs/msg/Int32` at 33 ms, RELIABLE / VOLATILE / KEEP_LAST(10), Data Sharing AUTOMATIC, and ROS loaning enabled. There are three persistent subscribers and a fourth repeatedly started/stopped. The subscriber uses a `const Message &` callback, so `UniquePtr` is not required to reproduce this failure. The runtime payload pool was confirmed to be `WriterPool`.

**Observed transition on the unpatched library**

Within one call to `check_acked_status()`, after the Reader traversal:

```text
WriterHistory contains DDS sequence 287
next_all_acked_notify_sequence_ = 287
Reader low marks = [287, 287, 287, 286]
Reader tracking-list sizes = [0, 0, 0, 0]
computed min_low_mark = 286
computed all_acked = true
```

The notification condition is false (`287 <= 286`). The later `all_acked` block then advances `next_all_acked_notify_sequence_` to 288, while DDS sequence 287 remains in WriterHistory and its shared payload remains unmarked for removal. I captured the before/after states and a core dump.

This is consistent with a separate exhausted-pool snapshot: WriterHistory retained DDS sequence 924, all four Readers had low mark 934 and empty tracking lists, and the notification cursor was already 935. The shared-history head was not marked removed, the following ten entries were marked removed, and the free-payload count was zero. `advance_till_first_non_removed()` stops at that head.

**Existing candidate and limited validation**

I found Tomoya Fujita's existing [candidate commit 2cb018d47](https://github.com/eProsima/Fast-DDS/commit/2cb018d47d25391085e47f80fc14b04c4f065ec7), which appears to address this mechanism. I applied its logic to v2.6.12, immediately after the Reader traversal and before notification processing:

```cpp
if (all_acked)
{
    min_low_mark = mp_history->next_sequence_number() - 1;
}
```

I built an unmodified baseline and the candidate with identical build options, selected each library through `LD_LIBRARY_PATH`, and verified the loaded library paths in the publisher's process maps.

| Trial | Unmodified v2.6.12 | With candidate |
|---|---|---|
| 1 | Borrow failures after 10.7 s | Completed 60 s without errors |
| 2 | Borrow failures after 7.6 s | Completed 60 s without errors |
| 3 | Completed 60 s without errors | Completed 60 s without errors |

Across the three candidate runs, 5,515 messages were published and 60 extra-subscriber start/stop cycles completed without borrow/publish errors or detected sequence gaps.

A small logic model also retained the history entry when an existing Reader genuinely had a pending change. That check used Reader/History substitutes; it was not a full DDS regression test. The candidate runs are short, and I have not captured the equivalent boundary transition on the patched library, so I would not consider this exhaustive validation.

Could you confirm whether this matches the intended fix in 2cb018d47, and whether a regression test/backport to the 2.6.x branch used by Humble would be appropriate?

fujitatomoya, 2026-11-05:
> Could you confirm whether this matches the intended fix in 2cb018d47, and whether a regression test/backport to the 2.6.x branch used by Humble would be appropriate?

looks like it. but i am not sure if that can be backported to downstream branch 2.6.x ... it is managed and maintained by eProsima. @MiguelCompany wdyt?