ros2/rmw_fastrtps #811: Set XCDR2 data representation in ROS2

bshantam97, 2025-03-10:
Hello, I am using XCDR2 to publish data in FastDDS 3. How do I configure ROS2 to accept XCDR2 serialization/de-serialization format? I believe ROS2 using XCDR1 encoding by default.

christophebedard, 2025-03-25:
You can configure Fast-DDS when used with ROS 2 through an XML configuration file: https://fast-dds.docs.eprosima.com/en/3.1.x/fastdds/ros2/ros2_configure.html#xml-configuration. I couldn't easily find a configuration for data representation, though. Maybe @juanlofer-eprosima knows more?

juanlofer-eprosima, 2025-03-26:
Hi @bshantam97, unfortunately this feature is currently unsupported and not in our roadmap. Please contact info@eprosima.com so we can hear about your use case and propose a valid solution.

fujitatomoya, 2025-04-03:
this is answered as not supported, contact to eProsima for feature development. i will go ahead to close this.

closed 2025-04-03 by fujitatomoya