ros2/rmw_fastrtps #812: After booting, there are cases where a specific topic does not communicate between certain nodes.

kbs907, 2025-03-11:
As stated in the title, everything usually works fine, but sometimes after booting, the topic is being published correctly (verified using ros2 topic echo), yet a specific node fails to receive the topic.

Additionally, a peculiar point is that the nodes that should publish the topics (like teleop-twist-joy or rectify image nodes) do not publish cmd_vel or rectify image even though the joy or camera topics are being published.

Since this was originally functioning normally, I believe there are no issues with the DDS configuration XML file, QoS settings, or ROS domain settings. Running ros2 doctor also doesn't show any significant problems.

I want to provide as much information as possible to identify the issue, but I'm unsure what is needed. If the problem occurs again, what methods can I use to identify the cause and provide you with relevant information?

The environment in which I am running the ROS 2 node is as follows:

- ROS 2 Humble
- 10 system services (5 running individual launch files, 4 running bag record scripts, 1 running Docker Compose)
- 3 Docker instances running
- Approximately 32 nodes are running in total
- There are 207 topics
- The ROS domain ID is unified as 11 (both host and Docker)

The Fast DDS configuration XML file that is loaded when each program is executed is as follows:
`<?xml version="1.0" encoding="UTF-8"?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>shm_transport</transport_id>
                <type>SHM</type>
                <segment_size>78905344</segment_size>         
            </transport_descriptor>
        </transport_descriptors>
        <participant profile_name="SHMParticipant" is_default_profile="true">
            <rtps>
                <userTransports>
                    <transport_id>shm_transport</transport_id>
                </userTransports>
                <sendSocketBufferSize>12582912</sendSocketBufferSize>
                <listenSocketBufferSize>12582912</listenSocketBufferSize>
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
        <data_writer profile_name="default publisher profile" is_default_profile="true">
            <qos>
                <data_sharing>
                    <kind>AUTOMATIC</kind>
                </data_sharing>
            </qos>
            <historyMemoryPolicy>DYNAMIC</historyMemoryPolicy>
        </data_writer>
        <data_reader profile_name="default subscription profile" is_default_profile="true">
            <qos>
                <data_sharing>
                    <kind>AUTOMATIC</kind>
                </data_sharing>
            </qos>
            <historyMemoryPolicy>DYNAMIC</historyMemoryPolicy>
        </data_reader>
    </profiles>
</dds>
`

The Docker Compose file is as follows:
`version: "3"
name: "t2"
x-proj-version: "t2-v2-ros2"
x-proj-updated: ${PROJECT_UPDATE_TIME}
x-robot-name: ${MODEL_NAME}

volumes:
   thira-configs-ros2:
     name: th-configs-ros2
     external: true
   mw-manager-param:
[... 168 lines cut ...]
      - "/dev/shm:/dev/shm:rw"
      - "/opt/thbot/docker-ros2/data/:/opt/thbot/params"
      - mw-manager-param:/home/th/middleware/middleware_manager/share/middleware_manager/param
      - mw-config:/home/th/middleware/middleware_config/share/middleware_config/config
      - mw-manager-node-config:/home/th/middleware/middleware_manager/share/middleware_manager/config/node_config
      - navi-manager-param:/home/th/middleware/navigation_manager/share/navigation_manager/params
      - mw-colcon-ws:/home/th/colcon-ws
      - mw-mqtt-config:/home/th/middleware/mqtt_client/share/mqtt_client/config
      - mw-data-store:/home/th/middleware/middleware_config/share/middleware_config/data
      - mw-interface-bridge:/home/th/middleware/interface_bridge/share/interface_bridge/config
      - mw-last-node:/home/th/middleware/state_manager/share/state_manager/data
     environment:
      - TZ=Asia/Seoul
      - ROS_LOCALHOST_ONLY=1
      - ROS_DOMAIN_ID=11
      - MIDDLEWARE_AUTO_START=true
      - ACTIVE_SETUP_UI=true
      - RMW_IMPLEMENTATION=rmw_fastrtps_cpp
      - RMW_FASTRTPS_USE_QOS_FROM_XML=1
      - FASTRTPS_DEFAULT_PROFILES_FILE=/opt/thbot/params/fastdds_shm_zero_copy.xml
     env_file:
      - /opt/thbot/Config/time.env
      - /opt/thbot/Config/get_ros2_log_path.env
     privileged: true
        #command: ["/bin/bash","-c","source /home/th/middleware/setup.bash; ros2 launch middleware_manager middleware.launch.py"]`

juanlofer-eprosima, 2025-03-11:
Hi @kbs907, thanks for the detailed report. At first sight this seems like a shared-memory related issue. It is important that your containers are stopped in a tidy manner, so that nodes (and their inner DDS entities) are removed without leaking allocated resources in the system. Otherwise, residual files might remain in the `/dev/shm` directory, which can cause the behavior you are describing.

I suggest you check if this is precisely what is happening to you, by inspecting this folder before launching your containers. If that is the case you may prevent the issue with a tidy closure, or perform a manual cleanup just before launching your setup. Perhaps you could also try to substitute SHM for UDP just to confirm it is a shared memory related problem.

kbs907, 2025-03-14:
@juanlofer-eprosima 
Thank you. I will check as you mentioned. However, when problems occur, I check the usage of /dev/shm using the df -h command, and it shows about 30% usage. Could this still be an issue?

fujitatomoya, 2025-04-03:
@kbs907 any update? if you do not see any problem with suggested approach, please close this issue.

rty813, 2025-04-09:
@fujitatomoya When the host is rebooted, won't /dev/shm be automatically cleared?

fujitatomoya, 2025-04-09:
@rty813 i am not certain yet, but i think so. it seems to depend on the main process in the container.

rty813, 2025-04-10:
I proactively created a blank file in the /dev/shm directory, and the file disappeared automatically after the reboot. So is it certain that the environment will be clean after the reboot?

fujitatomoya, 2025-04-10:
@rty813 what do you mean by this `reboot`? if that is the host system, of course it cleans everything.

rty813, 2025-04-12:
Since /dev/shm is cleaned up when the host reboot, why is there still a situation of _"leaving the allocated resources in the file system"_?

kbs907, 2025-04-25:
The issue was caused by running a ROS 2 bag record without loading the DDS settings from one of the systemctl services. 
After disabling that service, the problem did not occur. 
Thank you!

closed 2025-04-25 by kbs907