ros2/rmw_fastrtps #815: FastDDS Publisher ROS2 Subscriber Topic Type Hash

bshantam97, 2025-03-17:
Hi, I am publishing MarkerArray Data from fastdds and ROS2 is subscribing to that. I can see ROS2 discovers the topic but when I perform a ros2 topic info i get a Topic Type Hash: INVALID. Under what circumstances does this take place ?
I am using ROS2 rolling and FastDDS3 RMW.

juanlofer-eprosima, 2025-03-20:
Hi @bshantam97, would you be able to provide a reproducer? Or at least describe better your setup specifying the exact ROS 2 and Fast-DDS versions used in each component. It is expected that the topic type hash is not sent from a pure Fast-DDS application, however this should not be affecting matching or data reception.

bshantam97, 2025-03-20:
Hi @juanlofer-eprosima thank you for your response. So i currently reverted back to fastdds 2.13 with ros2 humble and everything works as expected. Previously I used FastDDS 3.0.1 with the ROS2 rolling branch. It seems the data matching happened but On the ROS2 subscriber side aka rviz it was unable to decode the MarkerArray Topic as the hash is invalid.

juanlofer-eprosima, 2025-03-21:
Hi @bshantam97, without a reproducer it will be difficult for us to help. Which IDL is your Fast-DDS publisher app using? If it's the one generated by ROS 2 msg to idl converter, that could be the problem and you might need to change it slightly. We are currently preparing a tutorial to deal with this scenario. I can also suggest to try with the latest Fast-DDS release (v3.1.2) just in case.