ros2/rmw_fastrtps #820: SHMEM Transport errors: open_and_lock_file failed

rty813, 2025-04-08:
### Operating System:

Ubuntu arm64

### ROS version or commit hash:

Humble

### RMW implementation (if applicable):

rmw_fastrtps_cpp

### Client library (if applicable):

rclpy

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,UP,RUNNING>
mtu          : 65536
inet         : 192.168.3.112
inet4        : ['192.168.3.112']
ether        : 48:b0:2d:eb:5c:e2
netmask      : 255.255.255.0
device       : eth0
flags        : 4163<MULTICAST,UP,RUNNING,BROADCAST>
mtu          : 1500
broadcast    : 192.168.3.255
inet         : 10.76.81.191
inet4        : ['10.76.81.191']
ether        : da:d8:cd:a6:45:00
netmask      : 255.255.255.128
device       : usb0
flags        : 193<NOARP,UP,RUNNING>
mtu          : 1430
inet         : 192.168.55.1
inet4        : ['192.168.55.1']
ether        : f2:a5:f7:17:01:5d
netmask      : 255.255.255.0
device       : l4tbr0
flags        : 4099<MULTICAST,UP,BROADCAST>
mtu          : 1500
broadcast    : 192.168.55.255
inet         : 10.1.123.79
inet4        : ['10.1.123.79']
ether        : 72:2d:46:2a:a6:bc
netmask      : 255.255.0.0
device       : edge0
flags        : 4163<MULTICAST,UP,RUNNING,BROADCAST>
mtu          : 1290
broadcast    : 10.1.255.255
inet         : 172.17.0.1
inet4        : ['172.17.0.1']
ether        : 02:42:d2:90:71:a1
netmask      : 255.255.0.0
device       : docker0
flags        : 4099<MULTICAST,UP,BROADCAST>
mtu          : 1500
broadcast    : 172.17.255.255
inet         : 172.18.0.1
inet4        : ['172.18.0.1']
ether        : 02:42:8d:d3:07:51
netmask      : 255.255.0.0
device       : br-51228a570487
flags        : 4163<MULTICAST,UP,RUNNING,BROADCAST>
mtu          : 1500
broadcast    : 172.18.255.255
ether        : c6:ac:50:6e:be:d8
device       : dummy0
flags        : 130<NOARP,BROADCAST>
mtu          : 1500
ether        : f2:a5:f7:17:01:5d
device       : rndis0
flags        : 4099<MULTICAST,UP,BROADCAST>
mtu          : 1500
ether        : f2:a5:f7:17:01:5f
device       : usb1
flags        : 4099<MULTICAST,UP,BROADCAST>
mtu          : 1500
ether        : 8e:83:d7:0c:4b:6b
device       : veth5d02208
flags        : 4163<MULTICAST,UP,RUNNING,BROADCAST>
mtu          : 1500

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.10.104-tegra-aarch64-with-glibc2.29
release          : 5.10.104-tegra
processor        : aarch64
```
</details>

### Steps to reproduce issue

I started multiple nodes, and after running normally for a while, some topic data suddenly stopped. 

### Expected behavior

everything ok

### Actual behavior

When checking the message frequency with ros topic hz, an error occurred: 
`[RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7425: open_and_lock_file failed -> Function open_port_internal`
Then, I checked /dev/shm and found only "fastrtps_port7425_el" but no "fastrtps_port7425". After deleting all files in /dev/shm and restarting the process, it recovered.

fujitatomoya, 2025-04-08:
Do you happen to use docker container?

rty813, 2025-04-08:
Yes, it runs in a docker container. But I set some settings related to shared memory in docker-compose:
```
network_mode: host
ipc: host
uts: host
privileged: true
container_name: prd_foxglove
restart: always
ulimits:
  core: 0
```

fujitatomoya, 2025-04-08:
https://github.com/ros2/rmw_fastrtps/issues/812#issuecomment-2759353346 could be similar issue.