ros2/rmw_fastrtps #821: [Question] Discovery behavioral difference between `humble` and `rolling` with localhost_only

fujitatomoya, 2025-05-03:
### Description

This is not the bug report, but it is more like question why this is happening for discovery.
When the localhost_only is enabled either with `export ROS_AUTOMATIC_DISCOVERY_RANGE=LOCALHOST` and `export ROS_LOCALHOST_ONLY=1`, the discovery behavior is different between `humble` and `rolling`.

### Details

- `rolling`

  Terminal-2 is not enabled localhost only, but it still can discover the component_container in the same local system. this means that localhost only setting is not exclusive if they are in the same localhost. 

```console
### Terminal-1 (only this node is localhost only setting enabled)
root@tomoyafujita:~/ros2_ws/colcon_ws# export ROS_AUTOMATIC_DISCOVERY_RANGE=LOCALHOST
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run rclcpp_components component_container

### Terminal-2 (localhost only is disabled)
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon stop
The daemon is not running
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list --no-daemon
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list --no-daemon --spin-time 3
/ComponentManager
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon status
The daemon is not running
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 daemon start
The daemon has been started
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list
/ComponentManager
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list --no-daemon
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 component list --no-daemon --spin-time 3
/ComponentManager
```

on the other hand,

- `humble`

  Terminal-2 is not enabled localhost only, with humble it CANNOT discover the component_container in the same local system. this means that localhost only setting is exclusive, even if they are in the same localhost system. 

```console
### Terminal-1 (only this node is localhost only setting enabled)
root@tomoyafujita:~# export ROS_LOCALHOST_ONLY=1
root@tomoyafujita:~# ros2 run rclcpp_components component_container

### Terminal-2 (localhost only is disabled)
```shell
root@tomoyafujita:~# ros2 daemon stop
The daemon is not running
root@tomoyafujita:~# ros2 component list --no-daemon
root@tomoyafujita:~# ros2 component list --no-daemon --spin-time 3
root@tomoyafujita:~# ros2 daemon status
The daemon is not running
root@tomoyafujita:~# ros2 daemon start
The daemon has been started
root@tomoyafujita:~# ros2 component list
root@tomoyafujita:~# ros2 component list --no-daemon
root@tomoyafujita:~# ros2 component list --no-daemon --spin-time 3
```

fujitatomoya, 2025-05-03:
@MiguelCompany do you have any idea about this?

fujitatomoya, 2025-05-03:
CC: @Barry-Xu-2018

juanlofer-eprosima, 2025-05-06:
@fujitatomoya My guess is this has to do with an old issue fixed [here](https://github.com/eProsima/Fast-DDS/pull/3733). Could you please check that by exporting `ROS_LOCALHOST_ONLY` in terminal 2 you see the same behaviour as in rolling?

fujitatomoya, 2025-05-06:
> Could you please check that by exporting ROS_LOCALHOST_ONLY in terminal 2 you see the same behaviour as in rolling?

yeah, if we set `ROS_LOCALHOST_ONLY` in terminal 2, everything works okay. my question was that even if `ROS_LOCALHOST_ONLY` is set only on the specific process space, the participant in the same localhost system without `ROS_LOCALHOST_ONLY` should be able to discover that participant as well. (this behavior only works with rolling, kilted and jazzy but humble. that was the original question.)

https://github.com/eProsima/Fast-DDS/pull/3869 did not make it to `2.6.x` branch because of the conflicts, so this is not gonna be fixed in humble.

I understand the reason and situation, so i will go ahead to close this issue. thank you very much for sharing the information.

closed 2025-05-06 by fujitatomoya