ros2/rmw_fastrtps #822: minimal_publisher_with_wait_for_all_acked: Not all subscribers acknowledge

mjcarroll, 2025-05-09:
### Operating System:

ghcr.io/sloretz/ros-testing:kilted-desktop

### ROS version or commit hash:

kilted

### RMW implementation (if applicable):

rmw_fastrtps

### Steps to reproduce issue

https://docs.ros.org/en/kilted/Tutorials/Demos/Wait-for-Acknowledgment.html

### Expected behavior

All messages ack

### Actual behavior

Not all messages ack

### Additional information

Not all subscribers acknowledge messags when running `minimal_publisher_with_wait_for_all_acked`: 

https://github.com/ros2/kilted_tutorial_party/issues/542#issuecomment-2845988577

fujitatomoya, 2025-05-09:
@mjcarroll thanks for the issue. i will take a look at this.

fujitatomoya, 2025-05-09:
i think this is just a racy condition, after signal comes in it waits on the ack for 300msec, probably this is too short to wait. i tested with 1000msec, and it always confirms all subscribers to acknowledge the messages. to avoid this racy condition in the 1st place, i suggest that we could give it negative value for the timeout so that ack waits on forever until all subscribers acknowledge the messages. if subscriptions exit during running the publisher, this connectivity information is gonna be notified to publisher endpoint (DDSWriter), and publisher does not wait for the ack from non-existing subscription.

closed 2025-05-13 by fujitatomoya